matpi/EmbeddedSystemsUTU #10: Sous Vide -keitin (Anna Kohtamäki & Juuso Laine)

porilainentaparikollinen, 2015-01-13:
Projektinamme on toteuttaa Sous Vide -keitin.

Sous Vide tarkoittaa tyhjiöpakattujen ruoka-ainesten kypsentämistä vesihauteessa tarkassa jatkuvassa lämpötilassa. Menetelmän etu on kontrolloidussa prosessissa, jossa jokaiselle ruoka-aineelle voidaan valita optimaalinen kypsennyslämpötila ilman nesteen haihtumista. 

Keitin on yksinkertainen komponenteiltaan. Vesihauteena käytetään riisikeitintä, jonka vastustusten arvoja Arduino säätelee releen avulla. Arduinoon on yhdistetty lämpötunnistin, joka mittaa keittimessä olevan veden lämpötilaa. Veden lämpötila voidaan säätää 0,1-0,5 celsiusasteen tarkkuudella.

Robojuchen, 2015-01-15:
Tässä on erittäin mielenkiintoinen aihe ja ilahduttaisi että se on todellinen tuote todelliseen käyttöön. Miten korkean lämpötilan laitteen pitäisi saavuttaa? 

Olemme matin kanssa pohtineet projetin turvallisuus ja vastuukysymyksiä kun siinä joudutaan nopeasti verkkovirran kanssa tekemisiin. Lisäksi laitteen pitäisi olla luotettava että sitä voisi käyttää valvomatta. Nämä kysymykset ovat mielestäni ratkaistavissa mutta katsotaan asiaa.

Olen miettinyt voisiko kuumennusta toteuttaa peltier-elementeillä jotka toimivat 12v tasajänniteellä ja pystyisvät myös jäähdyttämään tarvittaessa!

Onko teidän kypsennin joku valmis laite vai mitä oli mielessä? Mien meinasitte mitata lämpöilan. Tässä projetissa joudutaankin sitte PID-säätämisen kiehtovaan maailmaan.

juhasaario, 2015-01-16:
Tommone ku TMP36 on iisein mahdollinen kikkula, mutta tutustukaapa tollaseen ku DS18B20. Siinä olis mojoa.

Robojuchen, 2015-01-16:
Ainakin peltieriä ja noita lämpösensoreita voi kokeilla ensi demolla. On mulla vaihtovirralle soveltuva telekineettisesti mukana mutta mietitään Matin kanssa voiko niitä antaa lapsille :)