matpi/EmbeddedSystemsUTU #11: Robottiauto

Pikuz, 2015-01-13:
Tarkoitus on tehdä robottiauto joka lukee ultraäänianturin kautta arvoja ja kääntyy niin pitkään kunnes estettä ei enään ole edessä. Ajattelimme valmistaa auton rungon osat laserilla muovista.

Työtä varten tarvitsemme:
2 servoa
ultraäänianturi
arduino uno/nano boardi
muovilevyä

Tarvitsemmeko muita osia?

Mikael&Johan

Robojuchen, 2015-01-15:
Tässä on hyvä projektin pohja. Motivoiva ja siitä hyvä että sen kehittäminen on mukavaa. Ei ole sellainen projekti jota tehdään ja tehdään ja lopulta toimii tai ei. Tätä voi rakentaa hyvin vaihe vaiheelta ja kehittää eteenpäin niin paljon kun kunnianhimoa ja mielenkiintoa riittää.

Mitetittävää:

Projekti on nyt sellaisenaan vielä liian yksinkertainen ja muistuttaa demoilla tehtyä peruskonseptia.
Miettikää mitä sensoreita tai toimintoja voisitte lisätä robottiin. Miten se voisi reagoida ympäristöönsä? Vai sisältäisikö se jotakin erityistä mekaniikkaa?

Pikuz, 2015-01-23:
Kehitimme ideaamme siten, että auto käynnistyy RFID-tunnistuksella, sekä kohdatessa esteen, se antaa äänimerkin ja katolle syttyy vilkkuvalo. Esteen pysyessä edessä, auto kääntyy ja jatkaa kulkuaan.

Ostoslista:
2 servoa
ultraäänianturi
arduino uno boardi
RFID-tunnistin
akku
ledivaloja
Sekä muovilevyä ja muita rungon materiaaleja ja tarvikkeita.Tarvitsemmeko muuta?

Robojuchen, 2015-01-23:
Oikein hyvä!  Tätä voi halutessaan kehittää vielä että jos auto kohtaa rfid-lätkällä merkitylle alueelle ajellessaan sen käytös muuttuu :) Mutta ensin perusasiat kuntoon. Teettekö oman shieldin unolle? suositeltavaa. Ostakaa tarpeeksi vahvat ympäripyörivät servot, kaikista pienimmät eivät ole tarpeeksi vahvat.