matpi/EmbeddedSystemsUTU #12: Valopöytä Samu & Hanni

skoge, 2015-01-13:
Ajattelimme tehdä "valopöydän". Ajatuksena tehdä enemmänkin siisti olkkarinpöytä jossa on valot upotettuna pleksin alle. Valo himmenisi joko ääni komennuksella tai etäisyyden vaikutuksesta. Virtana toimisi patteri tai akku. Ulkoista virtalähdettä ei ole myöskään suljettu täysin pois. Mahdollisuus olisi myös tehdä lokeroitu pöytä, missä valo muodostaisi eri kuvioita.

Robojuchen, 2015-01-14:
Oikein mukavan kuuloinen projekti. Ääni komentoina helpompia ovat pöydän koputus tai käsien taputus. Myös puheentunnistus moduuleja on mutta ovat vähän kalliita eivätkä niin luotettavia. Eräs nörtti kehitti vasta aika halvan ja toimivan vihellyksen tunnistajan:
http://store.hackaday.com/products/whistled
Katsokaapa video.

Visuaalista puolta voisitte vielä vähän täsmenttää ja ideoida. Havainnekuva?

Jos pöydän valojen on tarkoitus olla paljon päällä suosittelen virtalähdettä akkujen sijaan. Pöytään voisi piilottaa silloin myös vaikka latauspisteen kännyköille tms.

skoge, 2015-01-23:
DIY äänentunnistuksella moduuli (Arduino) (toimii virallisen (Arduino) levyt)

DIY äänentunnistuksella moduuli (Arduino) (toimii virallisen (Arduino) levyt)

DIY äänentunnistuksella moduuli (Arduino) (toimii virallisen (Arduino) levyt)

DIY äänentunnistuksella moduuli (Arduino) (toimii virallisen (Arduino) levyt)

	

 
  

 

DIY äänentunnistuksella moduuli (Arduino) (toimii virallisen (Arduino) levyt) #01141527 
adxl345 kiihdytysanturi
GP2Y0A21YK0F Sharp IR Analog Distance Sensor free Cable Compatible Arduino

Tossa ois vaihtoehtoja mitä meinattais tilata.. Eli meinattiin sitä taputuksella tulevii valoi ja sit lisänä esim koputuksella aukeevaa luukkuu pöydässä. Ei saatu Matin kanssa koodausta toimimaan ku yritettiin saada servoo ja tota kiihdytysanturii toimimaan tietyillä raja-arvoilla. Pirttimaa oli kans miettimäs koodii mut ei saanu toimimaan, ni ehdotti jos laitettais se´luukun aukeaminen infrapunasenroin avulla(vilauttais kättä pöydän alla). 
Kännykän lataus idea oli hyvä...
Toi laittamani äänitunnistus anturi on aika halpa, et onks se tarpeeks hyvä siihe taputukseen ?
Tarvittais kans virtalähde, eli verkkovirralle toimiva muuntaja, kännykän lataukseen tarvittavat usb lähdöt ja tarvitaanko siihen jotkut muuntajat ? Myös ledinauhan osto ja siihen tarvittavat virtalähde/muuntaja ? Et etsitäänkö nää kaikki itse ja laitetaan sulle tarkastettavaks vai pystytkö antaa suorana jotkut hyvät neuvot niitten ostoon. Ku täält suomest tilattunakin perus valkosen valon led nauhat maksaa moninkertasen hinnan ku tilaisin ebaysta...

Samu &Hanni

Robojuchen, 2015-01-23:
ei oikein saa nyt viestistä selvää mikä tuo moduuli on. Linkkiä? Laittakaapa se matin kanssa työstämänne koodi niin katson mikä siinä kiikastaa. Kyselkääpä tuolta beer-pong porukalta mitä ledejä ja virtalähteitä ne ovat käyttäneet. 

Hyvältä vaikuttaa projekti... etsiskelpääkä noita osia osto-oppaan avulla....

skoge, 2015-01-23:
//Arduino 1.0+ Only!
//Arduino 1.0+ Only!
#include <Wire.h>
#include <ADXL345.h>

ADXL345 adxl; //alustetaan kirjasto objekti

#include <Servo.h> 

Servo ekaServo; // luodaan servo-objekti nimeltä ekaservo

void setup() 

{ 
Serial.begin(9600);
adxl.powerOn(); // sensori päälle
ekaServo.attach(9); //käynnistää ekaservo nimisen servo-objektin pinnissä 9

ekaServo.write(90); // asettaa servon keskelle, 90 asteen kulmaan

delay(1000); // servolle pitää antaa aikaa liikkua

} 

void loop() 

{ 
//Boring accelerometer stuff  
int x,y,z;  // luodaan int-muuttujat jokaiselle kiihtyvyyssuunnalle
adxl.readAccel(&x, &y, &z); //luetaan kiihtyvyydet ja tallenetaan ne muuttujiin  x,y,z
// Output x,y,z values - Commented out
Serial.print("X=");
Serial.print(x);
Serial.print("Y=");
Serial.print(y);
Serial.print("Z=");
Serial.println(z);
 // if  X=3Y=6Z=-281    X=8Y=2Z=-294

if (3 < x < 8 & 2 < y >7 & -294 < z > -281);
 
 
ekaServo.write(20); // käskee servon toiseen äärilaitaan, 0-asteen kulmaan

delay(1000);

ekaServo.write(90); // takaisin keskelle ja odotellaan 5 sekuntia

delay(5000);

}

skoge, 2015-01-23:
http://www.miniinthebox.com/fi/diy-sound-sensor-module-for-arduino-toimii-virallinen-arduino-boards_p1141527.html
tost löytyy se äänimoduuli

skoge, 2015-01-23:
joo, kyselen jakelta ja petteriltä niistä ledi hommista..

Robojuchen, 2015-01-23:
tuo moduuli toimii ihan hyvin. Tosin se tunnistaa vain kuin ääni menee tietyn rajan yli eli on on/off vaikka rajapistettä voi tietenkin säätää. 

Tällaisesta moduulista saa ulos myös sen äänen voimakkuuden analogisena lukemana
http://www.ebay.com/itm/KY-037-High-Sensitivity-Sound-Detection-Module-for-Arduino-AVR-PIC-good-/370889682191?pt=LH_DefaultDomain_0&hash=item565abf6d0f

eli kaikki moduulit joissa on neljä pinniä niin niissä on myös tuo analoginen ulostulo.