matpi/EmbeddedSystemsUTU #7: Laser pallo, Juuso Vilponen, Aatu Vehosmaa ja Jere Niemi

juusovil, 2015-01-13:
Tarkoituksena toteuttaa Pienillä laser valoilla toimiva valopallo, jota voidaan kaukosäätimen avulla ohjata. Kaukosäätimellä pystytään säätelemään valojen rytmiä ja luomaan erilaisia valoefektejä. 
Komponentit tarkoitus mahduttaa tennispallon sisään.

Tilatut osat:
3.7V 850 mAh Rechargeable Polymer Li Battery For GPS DVD Camera Tablet PC 353562
Arduino compatible Nano V3.0 ATmega328 Mini USB Microcontroller Board Cable HIYP
10/20/50 pcs lot 650nm 6mm 3V 5mW Laser Dot Diode Module Red Copper Head
DC12V 6A Mini 44Key IR Remote Controller for 5050 3528 SMD RGB LED Strip Light
Sähköjohtoa

Lisätietoja:
http://www.instructables.com/id/Laser-Ball/?ALLSTEPS

[image]

juusovil, 2015-01-13:
Laservalo taroitus hajoittaa pienien prismojen avulla.

Robojuchen, 2015-01-14:
Hyvän oloinen projekti! Tästä saa aika makean esineen.

Hyviä puolia:

Simppeli ja kokeiltu peruskonsepti.
Kauko-ohjaus ja valojen ohjaus hyviä sisältöjä myös koululaisprojektia ajatellen.
Mielenkiinotinen tiivis toetutus tennispalloon
Tästä saa näyttävän ja mielenkiintoisen, eikä ole liian vaikea varmasti ohjelmoida.

Kehitettävää:
Mitä persoonallista lisäätte tähän hyvin valmiiseen konseptiin. Jokin kolmas elementti pitäisi saada.
Saisiko pallon reagoimaan johonkin.

esim.
Kiihtyvyysanturilla liikkeeseen
Mikrofonilla ääneen.
Liikkeeseen tai etäisyyteen.

 Oletteko miettineet tuota patterin lataamista? Arduino pro mini olisi muuten halpa ja tuota nanoa pienempi ja lisäksi vie vähemmän virtaa.

juusovil, 2015-01-29:
Päivitys:
Tilatut osat saapuneet,
-Arduino nano
-usb-johto
-laserit
-prisma
-akku
-kaukosäädin, ja johto

Aloitimme demolla prismpojen asentamisen lasereihin, ja saimme jokaiseen (28) asennettua. Tarkoituksena käyttää vain 14, joten poimimme parhaat päältä pois lopulliseen tuotteeseen.

Ideana lisätä tuotteeseen kolmas elementti kiihtyvyysanturilla (vielä tilaamatta). Kun palloa liikutetaan/pyöritetään, laser valojen rytmi muuttuu nopeammaksi.

Kumpaa kannattaa mielestäsi käyttää while- vai if-lauseketta koodissa?

Robojuchen, 2015-01-30:
Tuo lause riippuu ihan kummalla sen saa toeutettua helpommin. Molempia voi käyttää vaikka sekaisin ja yhtä aikaa :)

Tuo kiihtyvyysanturi on tosi hyvä idea, magic-efekti kun palloa vispaa, ooooh

juusovil, 2015-02-27:
Oliko nyt niin, että robojuchenilta löytyy arduino pro mini, jonka olisit valmis mymmänn edullisella opiskelijahinnalla meille? Mietittiin vaan, että jos löytyisi tätä kautta niin ei ole järkevää tilata merten takaa.

Robojuchen, 2015-02-28:
Voin myydä pro minin teille siihen hintaan mihin hintaan olen sen ostanut eli 3e. Teidän pitäis nyt selvittää tarvitsetteko 3.3v vai 5v version.

juusovil, 2015-03-27:
Nyt Arduino jaksaa valittaa. Viime kerralla katsottiin mielestäni porukassa koodi kuntoon, mutta kun tänään avasin koodin uudestaan, alkoi ohjelma varoittaa.

Tässä koodi:
#include <IRremote.h>
#include <Wire.h>
#include <ADXL345.h>

ADXL345 adxl; //alustetaan kirjasto objekti

//IR remote globals
int RECV_PIN = 2;
IRrecv irrecv(RECV_PIN);
decode_results results;
int pat_select;

//Interrupt globals
int int0 = 0; //interrupt 0 is on pin 5 on the Teensy;
volatile int state = LOW; //'volatile' is necessary for interrupts

//Lasers globals
int wait = 100;
int n_lasers = 14;
int laser[] = {3,4,5,6,7,8,9,10,11,12,13}; //remaps the Teensy pin numbers to consecutive laser numbers

void setup()
{
  //IR setup
  Serial.begin(9600);
    adxl.powerOn(); // sensori päälle
  irrecv.enableIRIn(); // Start the receiver
  pat_select = 0;
  
  //Interrupt setup
  attachInterrupt(int0,stateChange,CHANGE);
  
  //Laser setups
  for (int i=0;i<n_lasers;i++)
  {
    pinMode(laser[i],OUTPUT);
  }
}
  

void loop() 
{
  
    int x,y,z;  // luodaan int-muuttujat jokaiselle kiihtyvyyssuunnalle
  adxl.readAccel(&x, &y, &z); //luetaan kiihtyvyydet ja tallenetaan ne muuttujiin  x,y,z

  // Output x,y,z values - Commented out
  Serial.print("X=");
  Serial.print(x);
  Serial.print("Y=");
  Serial.print(y);
   Serial.print("Z=");
  Serial.println(z);

  laser_patterns();
  
}

//Interrupt function
void stateChange(){
  if (irrecv.decode(&results)) { //decode incoming IR signal and store in 'results'
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  
  switch(results.value){  
      case 0xFD00FF:
        Serial.println("VOL-"); //useful for debug
        if (wait>25) wait-=25;
        results.value = 0x00;
      break;
      case 0xFD40BF:
        Serial.println("VOL+");
        wait+=100;
        results.value = 0x00;
      break;
      case 0xFDA05F:
        Serial.println("UP");
        pat_select++;
        results.value = 0x00;
      break;
      case 0xFDB04F:
        Serial.println("DOWN");
        if (pat_select>0) pat_select--;
        results.value = 0x00;
      break;
          
  }
  state = !state;
}

void laser_patterns(){
  switch (pat_select){
    case 0: //All-on
      for (int i=0;i<(n_lasers);i++){
        digitalWrite(laser[i],HIGH);
        digitalWrite(laser[n_lasers-1-i],HIGH);
        delay(wait);
      }
    break;
    case 1: //Bounce
      for (int i=0;i<n_lasers;i++){
        digitalWrite(laser[i],HIGH);
        delay(wait);
      }
      for (int i=0;i<n_lasers;i++){
        digitalWrite(laser[i],LOW);
        delay(wait);
      }
    break;
    case 2: //Dual-bounce
      for (int i=0;i<n_lasers-1;i++){
        digitalWrite(laser[i],HIGH);
        delay(wait);
        digitalWrite(laser[i],LOW);
        delay(wait);
      }
      for (int i=n_lasers-1;i>0;i--){
        digitalWrite(laser[i],HIGH);
        delay(wait);
        digitalWrite(laser[i],LOW);
        delay(wait);
      }
    break;
    case 3: //Wave
      for (int i=0;i<(n_lasers/2);i++){
        digitalWrite(laser[i],HIGH);
        digitalWrite(laser[n_lasers-1-i],HIGH);
        delay(wait);
        digitalWrite(laser[i],LOW);
        digitalWrite(laser[n_lasers-1-i],LOW);
        delay(wait);
      }
    break;
  }
  

}

Ja näistä ohjelma valittaa:

Laser_Ball_koodi:30: error: 'IRrecv' does not name a type
Laser_Ball_koodi:31: error: 'decode_results' does not name a type
Laser_Ball_koodi.ino: In function 'void setup()':
Laser_Ball_koodi:48: error: 'irrecv' was not declared in this scope
Laser_Ball_koodi.ino: In function 'void stateChange()':
Laser_Ball_koodi:86: error: 'irrecv' was not declared in this scope
Laser_Ball_koodi:86: error: 'results' was not declared in this scope
Laser_Ball_koodi:91: error: 'results' was not declared in this scope

Ideoita?

Robojuchen, 2015-03-30:
Teidän koodista ei näe nyt onko nuo includet kohdallaan. Laittakaa liitetidostona koodi ja pallon alkuperäinen koodi.

juusovil, 2015-04-27:
koodia ladatessa tulee erroriksi "Error compiling". Mistä on kyse?

juusovil, 2015-04-27:
C:\Program Files (x86)\arduino_163\libraries\RobotIRremote\src\IRremoteTools.cpp:5:16: error: 'TKD2' was not declared in this scope
 int RECV_PIN = TKD2; // the pin the IR receiver is connected to

jerenie, 2015-04-29:
Meidän blogi http://laserpallo.blogspot.fi/

Robojuchen, 2015-04-29:
laittakaapa linkkiä noihin ohjeisiin joilla olette yrittäneet korjata ongelmaa. Kun teidän koodi kääntyi sillä yhdellä koneella demolla niin mikä on muuttunut sen jälkeen? Onko koodia muutettu? Onko konetta vaihdettu? Onko arduino ohjelmaa päivitetty. 

Vaikuttaisi että siellä on joku kirjasto jossa tuo RECV_PIN on ,määritelty arduinon kannalta väärin. JOs ette kyseistä kirjastoa tarvitse sen voi poistaa libraries kansiosta.

jerenie, 2015-04-29:
http://www.instructables.com/id/Remote-Control-Light-switch-ARDUINO/
Tuolla sivulla oli samanlaista ongelmaa. Emme voineet poistaa RobotIRremotea, koska meillä ei ole käyttöoikeuksia.

Robojuchen, 2015-04-29:
kokeilitteko sillä koneella millä se toimi aiemminki? Kysykää matilta oikeuksia poistaa se kirjasto, mutta kyllä niitä pitäisi pystyä muokkailemaan. Pitää selvittää asiaa. Jos löydätte joltakin koneelta vanhemman version aruinosta esim. arduino 1.0.5 niin siinä ei pitäisi olla tuota häiritsevää kirjastoa