matpi/EmbeddedSystemsUTU #8: Peittoa vetävä herätyskello - Ari Rantanen

arperan, 2015-01-13:
Tavoitteenani on valmistaa herätyskello, joka vetää peiton päältä herättäessä.

Toiminta-ajatuksia:   

-Arduino 
-Lcd näytön avulla ajastettava herätyskello
-Servoilla toimiva kela, josta jokin ohut naru/vaijeri sidotaan vahvalla hakaneulalla peittoon
-Muutamia toimintoja: esim. raakaherätys (nopeasti peitto pois), ärsyttävä herätys (hitaasti peitto pois)

Kysymyksiä: 

Voiko servon asettaa ns. vapaalle, että kelan voisi esim manuaalisesti kelata takaisin herätystoimenpiteen jälkeen?

Robojuchen, 2015-01-15:
Tämä on inspiroiva projekti. Keskiössä ovat seuraavat kysymykset:

Näytön ja näppäimistön valinta (aika pitkälle esteettinen kysymys)
Peitonvetäjän moottori ja mekaniikka, varsinkin takaisinvedon ratkaisu.

Sevoa eikä vaihteistomoottoreita voi oikein heittää vapaalle. Oikeastaan mikä tahansa tarpeeksi voimakkaasti vääntävä moottori pistää liikaa vastaan. Ongelma olisi ratkaistava muutoin.

Hobbyking.com, sieltä löytyy isoja sailwinch-servoja joissa riittää vääntöä. DC-vaihteistomoottorit voisivat olla ketävämpiä ja hiljaisempia. Mekaniikasta pitäsi alkaa kehittelemään hahmotelmaa.

arperan, 2015-02-27:
Dear Jussi! Vihdoin pääsee tenttikirjojen ääreltä oikeisiin hommiin ja yritän laittaa tätäkin projektia nyt vähän eteenpäin.

[image]
Tuossa vähän pöllittyä vuokaaviota. Minusta tuo on suht selkeä ja tuon mukaan voisi vaikka edetä ?
Kannattaako mun kopioida joku rtc koodi johon liitän omat tarpeeni? Vai miten saisin koodihomman parhaiten käyntiin? (alottamisen vaikeus :)

[image]
Tässä raakaversiota valmiista tuotteesta. Kelan ratkaisussa pysyttelen ihan manuaalimeiningissä. Eli naru tms. pyöritetään itse haluttuun etäisyyteen ja kela vain sitten vetää sen takaisin herätyksen yhteydessä. Tuon kelan yritän sorvata alumiinista metallisorvilla. Muun tuotteen valmistan vanerista ja laatikko toimisi myös tuollaisena mini yöpöytänä esim. silmälaseille tai kirjalle. Laatikko tulee olemaan suht reilun kokoinen, joten elektroniikalle on hyvin tilaa.

Tässä kauppalistaa, voitko tsekata sen ja kommentoida onko ok laittaa tuollaiset tilaukset tulemaan?

Virtalähde: löytyy itseltä

Arduino Uno: http://www.ebay.com/itm/UNO-2011-ATmega328P-PU-microcontroller-Funduino-Arduino-Compatibl-IDE-USB-cable-/121074071940?pt=LH_DefaultDomain_0&hash=item1c3093b984

Rtc: http://www.ebay.com/itm/DS3231-AT24C32-IIC-module-precision-Real-time-clock-module-memory-module-Arduino-/400503978923?pt=LH_DefaultDomain_0&hash=item5d3fe5bbab

Buzzer: http://www.ebay.com/itm/2PCS-5V-Active-Buzzer-Magnetic-Long-Continous-Beep-Tone-Alarm-Ringer-12MM-/400814713029?pt=LH_DefaultDomain_0&hash=item5d526b28c5

Periaatteessa tuohon herätysääneen voisin panostaa jollain muullakin kun tolla piipperillä. Mietin onko toi liian hiljanen ääni? Toisaalta toi piippauslaulujen säveltäminen vois olla hauskaa.

Ltc/näppikset: http://www.ebay.com/itm/1602-LCD-Board-Keypad-Shield-Blue-Backlight-Arduino-LCD-Duemilanove-Robot-/170928504704?pt=LH_DefaultDomain_0&hash=item27cc21ff80

Dc moottori: http://www.ebay.com/itm/DC6V-12V-24V-High-torque-Turbo-worm-Geared-motor-DC-motor-JGY370-Metal-gear-/261592254465?pt=Apparel_Merchandise&var=&hash=item3ce81d0801&vxp=mtr

Mitäs tuohon moottoriin ny kannatikaan valita kierrosluvuksi?

Mitä muuta pitää ostaa, piuhoja ? Teen vissii lopulliseen settiin sitten piirilevyt yms. kiinteet juotokset?
Niihin taitaa löytyä koululta tarvittavat matskut.

Kiitos t. Ari

Robojuchen, 2015-02-28:
Erittäin hyvin laadittu tuo konsepit nyt. Tuo vuokaavio on fiksu homma, sen voi oikeastaan suoraan muuttaa koodiksi kun se on noin perusteellinen ja loogisesti ehjä.

Alat vain tutustumaan johonkin RTC kirjastoon ja sen esimerkkeihin. Toisaalta voit alkaa muuttaamaan tuota kaaviota koodiksi siten että käytät pseudokomentoja haeaika(); if (aikaAsetettu == true) jne. Toisin sanoen rakentamaan loogista rakennetta if ja while-lausekkeista ja aliohjelmista.

Valitut komponentit ovat päteviä. Tuo ääntelyelementti on pelkkä summeri, eli sillä ei voi vaihtaa sävelkorkeutta. Leppäkerttuprojektissakohan oli esitelty niitä piezolevyjä.

Tuohon vuokaavioon voisi lisätä vaihtoehdon että moottoria ei käytetä ollenkaan

Moottorin valintaan käytä myyjän tarjoamaa taulukkoa. Siinä on eri moottoreiden jännitteet(mikä on sinun virtalähteen jännite?) ja paljonko ne vie enimmillään virtaa ja pystyvät vääntämään. Suosittelisin ehkä jompaa kumpaa noista hitaimmista välityksistä vaikka vauhti onkin aika hiljainen. Laske tarvittava peiton nopeus tuon pyörimisnopeuden ja suunnitellun kelan ympärysmitan avulla.

Hyvä idea on tehdä tuo kela alumiinista että kestää. Tuo akseli on 5mm halkaisijaltaan. Voisi olla hyvä idea ettei yritä tehdä akselin muotoista sovitetta vaan lukitsee akselin kelaan tapilla/ruuvilla joka lävistää sivuttain kelan ja akselin .