matpi/EmbeddedSystemsUTU #9: Tutka sählymaaliin (tai kastelujärjestelmä parvekkeelle) Olli Vaaramo & Otto Hintsala

olpana, 2015-01-13:
Ajatuksena olisi tehdä tutka sählymaaliin. Tutka mittaisi laukauksen nopeuden ja ilmoittaisi sen näytöllä vetäjälle. 

Mikäli tutka-projekti osoittautuu kelvottomaksi, niin toisena vaihtoehtona Ollilla on tehdä automaattinen kastelujärjestelmä parvekkeelle. Kastelujärjestelmä annostelisi veden määrätyin väliajoin suoraan ruukkuihin. Viljeltäviä kasveja ovat mm. tomaatti, paprika, kurkku...

Apua tarvitsisimme tässä vaiheessa siihen, että mitä osia pitäisi tilata netistä projekteja varten.

Robojuchen, 2015-01-15:
Minulla on tuo tutkan piirilevy jo valmiina prototyyppiä varten, kunhan ehdin juottamaan sen ja testaamaan. Siinäkin tapauksessa että sen saa toimimaan alustavasti, ei voi oikein olla varma miten toimiva ja kestävä lopullinen laite on. Mutta erittäin mielenkiintoinen projetki joka sisältäisi myös makeaa pintaliitosteknologiaa yms hitec-hommaa.

Voisitte kuitenkin varalta kehitellä hieman tuota kastelujärjestelmän perusteita..

juhasaario, 2015-01-16:
Jos kastelujärjestelmään päädytte, niin vanhasta pesukoneesa magneettiventtiili. Helppo ohjata, kestää verkostopaineen, omaa oikean kierteen ja on ilmainen.

olpana, 2015-01-29:
Nyt odotellaan Jussin kokemuksia mikroaaltotutkan toimivuudesta, eli saitko sitä juotettua/testattua?

Robojuchen, 2015-01-29:
Tässä parhaillaan alan testailemaan, sain juotettua. Huomasin vain että puukolla poistamani tinasilta olikin ihan tuikitarpeellinen johdin, tekevälle sattuu...

Robojuchen, 2015-01-29:
Ihme kyllä vempele toimi! Sain isompaa lukemaa kun nyrkkeilin kohti laitetta vs. kun liikutin hitaasti kättä. Pitää vielä vähän tutkia miten tarkasti tuo nyt pelittää ja saadaanko siitä ihan kunnon nopeutta laskettua mutta näyttäisi että jonkinlaisen tutkan pystyy tekemään.

Robojuchen, 2015-01-30:
Saatte kokeilla tutkaa ensi demoilla. Tein vain vahingossa pienen laskuvirheen yhden komponentin arvossa ja siksi tämä versio tunnistaa nyt vain alle 20km/h nopeuksia mutta idea tulee silläkin selväksi.

olpana, 2015-02-12:
Huomenta Jussi! Jäi viimeviikolla se suodattimien vaihto tekemättä.. :/ Milloin olet tulossa Raumalle tutkasi kanssa? Pitäisi varmaan pian saada piirilevyt ja muut komponentit tilattua, jotta saadaan projektia eteenpäin.

Robojuchen, 2015-02-12:
http://www.limpkin.fr/index.php?post/2013/08/09/Making-the-electronics-for-a-%247-USD-doppler-motion-sensor

tuosta näette infoa tästä kytkennästä. Tuo ylärajataajuus määrittyy 1/(2 X pii X R6 X C6) ja 1/(2 X pii X R6 X C6) eli kaksi suodatinta on peräkkäin. nyt rajataajuus on 1/(2 x pii x 1megaohm x 0.68nanofaradia) tuo kun kerrotaan vielä tuhannella saadaan suurusluokka kohdalleen eli 234 Hz

laskin että jos juotatte noiden R6 ja R7 päälle eli rinnalle niin noin 75k vastukset tekevät toivotun säväyksen. eli tulee joku 160km/h ylärajataajuudeksi. jos poistatte ne kokonaan ensin ja juotatte uudet niin laittakaa vaikkapa 70k vastukset. Näytin oikeat vastukset matille. Ne ovat molemmin puolin mikropiiriä ja niissä lukee 105 eli 10+ viisi nollaa eli 1Mohm

Robojuchen, 2015-02-12:
tuo tinasilta kuvassa vasemmalla kondensaattorin ja sen 1Mohm vastuksen välillä pitäisi saada säilymään koska siinä oli pieni katkos kuparissa. Tilasin muuten jo ne piirilevyt tehtaalta. Toivottavasti saadaan ne nopeasti. Jännä homma kun en ole ennen tilannut tehtaalta.

Oletteko miettineet miten tuo tutkan lukema sitten ilmoitetaan ja mitä "toimintoja" laitteeseen tulee?

olpana, 2015-02-27:
Nonii nyt saatiin suodattimet vaihdettua.. Testaamaan emme taida nyt päästä, koska koodi on hukassa (ts. emme tallentaneet sitä viimeksi). :/ Joko piirilevyt kopsahti postilaatikosta? 
Joku näyttö tähän pitäisi liittää, jotta tulokset näkyisivät vetopaikalle saakka. Toimintoja voisivat olla esimerkiksi mahdollisuus kilpailuun kaveria vastaan tms, jolloin kamppeeseen nimettäis pelajaat jne.. (player1,player2 ...)? Se on mietityttänyt, että onkohan tuloksia mahdollista saada kännykän/padin näytölle, mikäli näytöt osottautuu liian hintaviksi meidän budjettiin.

Robojuchen, 2015-02-28:
Hieno homma, oliko hikinen puhde saada suodattimet vaihdettua? Ei ole piirilevyt edes lähteneet tehtaalta kun taisi tuo uusivuosi jumittaa nekin systeemit. 

Noihin kirjastoihin voitte tutustua tuolla:
https://www.pjrc.com/teensy/td_libs_FreqMeasure.html
https://www.pjrc.com/teensy/td_libs_FreqCount.html

Ovat asennettuna jo sille yhdelle koneelle.

Halvin näyttö olisi varmaankin bluetoothilla kännykän näytölle.

Robojuchen, 2015-03-30:
Sain tutkan toimimaan! Onnistuin mittaamaan sählymailalla huitomalla yli sadan kilometrin nopeuksia. Hyvin tunnisti myös tennispallon kun sitä heitti kohti tutkaa, harmi kun en löytänyt sählypalloa tähän hätään.

Tämä tutkamoduuli toimi ja ei antanut juuri yhtään häiriötä kun taas ne vihreät eivät oikein toimi suurilla nopeuksilla.

http://www.ebay.com/itm/ZY-JIN-HB100-Microwave-Sensor-10-525GHz-Doppler-Radar-Motion-Detector-Arduino-/251684795286?pt=LH_DefaultDomain_0&hash=item3a99953396

Siihen teidän näyttöön sopii joko pelkkä max7219 chippi jolle teette piirilevyn, tai sitten alla olevassa setissä on piirilevy valmiina.

http://www.ebay.com/itm/1pcs-MAX7219-Dot-matrix-module-MCU-control-Display-module-DIY-kits-/251672893572?pt=LH_DefaultDomain_0&hash=item3a98df9884

olpana, 2015-03-31:
Hienoa! Alkaa kuulostamaan siltä, että saadaan projekti onnistumaan. Saatiin just tilattua tuo tutkamoduuli sekä tuo chippi, molemmat noista linkeistä. Nyt pitäis alkaa seitsemän segmentin näyttöjä etsimään!

olpana, 2015-04-01:
Meillä on nyt tässä näitä näyttöjä ehdolla. Ei osattu päättää täällä demolla, että pitääköhän meillä olla yhteisanodi vai yhteiskatodi? Jos ehdit katsoa noita niin olisimme kiitollisia ja saatais tilaus tehtyä. :) Kolme 7-segmentin näyttöä tarvitsemme.

http://www.ebay.com/itm/5-Red-7-Segment-2-3-LED-Display-Digital-Tube-Common-Anode-1-Bit-Digitron-/141008250355?pt=LH_DefaultDomain_0&hash=item20d4bf25f3

http://www.ebay.com/itm/7-Segment-2-3-a-Common-Cathode-Digit-LED-display-pack-of-5-/281397111575?pt=LH_DefaultDomain_3&hash=item4184932b17

http://www.ebay.com/itm/2pcs-3inch-7-segment-LED-display-common-anode-/310768887517?pt=LH_DefaultDomain_0&hash=item485b44e6dd

http://www.ebay.com/itm/1-8-inch-7-segment-Blue-LED-display-common-anode-2pcs-/360616465901?pt=LH_DefaultDomain_0&hash=item53f66a95ed

Robojuchen, 2015-04-01:
Common cathode lukee ihan tuossa datasheetin toisella rivillä.

Robojuchen, 2015-04-01:
Jos tilasitte jo tuon max7219 piirin tai jonkin moduulin niin tilaisin teinä siihen varalta tuollaisen max7221 ic:n irtokappaleena.
http://www.ebay.com/itm/1PCS-MAX7221CNG-MAX7221-MAXIM-DRVR-DSPLY-LED-8DIG-24-DIP-NEW-GOOD-QUALITY-/201192258672?pt=LH_DefaultDomain_0&hash=item2ed7fe3c70

 Muuten sama piiri mutta tuottaa vähemmän elektronista häiriötä mikä voi olla tärkeää teidän projektissa.