matpi/EmbeddedSystemsUTU #13: Leppäkertturobotti

jarmoharri, 2015-01-14:
Tarkoituksena on siis rakentaa yksikertainen leppäkertturobotti, joka toimi ultraäänisensoriden avulla (tunnistaa esteen ja kääntyy/peruuttaa). Olemme muotoilleet tyhjiömuovauksella kotelon tarkoitusta varten. Lisäksi leppikselle tulee silmiksi RGB-ledit, joita voi ohjelmoida haluamallaan tavalla. 
Tarvikkeita yhteen robottiin, teemme tilauksen kuitenkin kaksinkertaisena ajatuksella molemmille omat leppikset.

1 ultraäänisensori
2 x servo
1 mikrokontrolleri
2 x RGB-ledi
6 x vastus 220 ohmia, RGB-ledejä varten
1 virtalähde/paristokotelo
1 kytkin on/off?

Kysymyksiä:
- minkälaiset servot ja ultraäänisensori kannattaisi valita?
- uno vai nano -mikrokontrolleri, kumpi parempi vai onko merkitystä?
- onko vastusten määrä ja ohmimäärä ok?
- paristokotelo riittänee?
- millainen kytkin?
- pystytkö suosittelemaan hankintapaikkaa?

harri&jarmo

Robojuchen, 2015-01-14:
Tämä on oikein hyvä projektin aihe! Hauska ja varmasti aika motivoiva kun saa ötökän mönkimään tahtonsa mukaan. On myös sovellettavissa suoraan koulu opetukseen.

Kehitettävää:
Tällaisenaan aika yksinkertainen ja konsepti kokeiltu ja opettettu jo demoilla. Nuo rgb-ledit tuovat vähän lisää sisältöä ja huolellisella viimeistelyllä tästä kompositiosta saa arvosanaksi hyvän.

Vositte vielä miettiä olisiko mahdollista että leppäkerttu reagoi vielä johonkin ärsykkeeseen ja/tai esim ääntelee tai muuten ilmaisee mielentilaansa. Nämä eivät ole vaikeita toteuttaa kuitenkaan. Onko muita ideoita millä tähän saisi omaperäisyyttä?

Servojen sijaan suosittelen vaihteistomoottoreita. Mikro kontrollerilla on väliä lähinnä hinnan ja robotin koon kannlta. Kytkimeksi suosittelen muita kuin kaikista pienimpiä moottoreiden vaatiman virran takia.

Koostan tässä yhteistä osto-opasta jonka julkaisen piakoin.

juhasaario, 2015-01-16:
Vaihteistomotteja hankkiessanne (jos niihin päädyte) kattokaa että mikä motor shield pitää hommata, että jaksaa pukata tarpeeksi virtaskaa.

jarmoharri, 2015-01-28:
Aiomme tilata yhtä robottia varten

1 ultraäänisensori
2 x servo	(DS04-NFC 360 Degree Servo Continuous Operation Steering Metal Gear Digital) Meidän taidoilla ei aleta vaihteistomoottoreja säätämään eli servot siis.               
1 mikrokontrolleri (arduino uno)
2 x RGB-ledi
6 x vastus 220 ohmia, RGB-ledejä varten
1 virtalähde/paristokotelo

Löysimme tällaisen äänijutskan ja ajattelimme hyödyntää tätä ideaa omassa työssämme.
https://learn.adafruit.com/adafruit-arduino-lesson-10-making-sounds/overview

piezo 
ldr (koululektroniikka)
vastus 1k ohm (koululektroniikka)

Millainen piezon tulisi olla? Tutkimme ebaytä, mutta emme tiedä mikä olisi sopiva.

Että näin.

Robojuchen, 2015-01-28:
Hyvältä vaikuttaa! Tuossa olis vaikka brittimyyjältä viiden pakkaus noita 35mm vois olla hyvä. Tuo #25 Maaginen silmä ryhmä tarvitsee myös piezo-elementtiä niin voisitte tilata yhdessä? 

http://www.ebay.com/itm/5-or-10-Piezo-Elements-Sounder-Sensor-Trigger-Drum-Accoustic-Disc-20mm-27mm-35mm-/360719728982?pt=UK_BOI_Electrical_Components_Supplies_ET&var=&hash=item53fc924156

Näitä voi myös löytää vanhojen tietokoneiden sisältä mustaan muovikoteloon koteloituna ja palovaroittimessa on myös pietsolevyjä, mutta niissä ei taida olla johtimia juotettuna.