matpi/EmbeddedSystemsUTU #14: Munakello - Sofia Krapi ja Siiri Knuuttila

somkra, 2015-01-14:
Haluaisimme toteuttaa ruuanlaitossa käytettävän munakellon. Ideana olisi siis, että siihen voisi asettaa nappia painamalla ajan, ja kun tämä aika on kulunut kuuluisi jokin äänimerkki.

Tarvitsisimme ainakin pienen arduino boardin (arduino nano?), numeronäytön, painonapin ja summerin.

Emme ole vielä hankkineet kaikkia tarvittavia osia, mutta tarkoitus olisi saada hankittua tarvittavat osat mahdollisimman pian, jotta päästään aloittamaan!

Robojuchen, 2015-01-14:
Todella hyvä projekti! Tähän saa kehitelyä oikein tyylikkään ja uniikin tuotteen. Nyt pitäisi hieman täsmentää ideaa ja mitä särmää tähän kelloon tulisi.

Ehdotan että toteuttaisitte käyttöliittymän ilman nappia kiihtyvyysanturilla! Kelloa koputtamalla ja kiertämällä asetetaan aika. Aikaa voi näyttää pinillä led-matriiseilla(led-matrix), seven segment näytöillä, pienellä oled näytöllä. Oled-näyttöä voin esitellä teille jos olette kiinnostuneet.

Tutustukaa noihin näyttöihin ja jatkakaa ideointia. Tuo nappi idea on myös hyvä jos haluatte satsata muuhun kuin käyttöliittymään. Tarvitsette projektiin arduino pro minin.

somkra, 2015-01-29:
Olemme miettineet osia, joita tarvitsisimme ja haluaisimme varmistaa sinulta vielä ennen ostamista. Eli ilmeisesti paras vaihtoehto olisi valita tuo arduino pro mini. Lisäksi haluamme käyttää pientä oled-näyttöä. Olemme päätyneet siihen, että haluamme säätää aikaa napin avulla ja siihen tarvitsemme siis jonkinlaisen painonapin. Vielä lisäksi summeri hälytystä varten. Tarvitsemmeko vielä jotakin olennaista näiden lisäksi?

Robojuchen, 2015-01-29:
suosittelen teille arduino pro mini 3.3v 8mhz versiota niin voitte ruokkia laitteen yhdellä 3.3v tai 3.7v patterilla. http://www.ebay.com/itm/Redesign-Pro-Mini-atmega328-3-3V-8M-Replace-ATmega128-Arduino-Compatible-Nano-/200914924969?pt=LH_DefaultDomain_0&hash=item2ec77675a9

Eli tuo paristotyyppi pitäisi päättää. Kuinka pienen ja sievän haluatte laitteesta? Nyt pitäisi hahmotella vähän tuota kotelointia.

Jos projektissa muuten käytetään vain nappia ja näyttöä niin voisitte satsata tuohon pieneen kokoon ja virransäästöominaisuuksiin yms. Ts. tehdä mahdollisimman valmiin tuotteen. 

http://www.ebay.com/itm/CRIUS-CO-16-OLED-Display-Module-V1-2-for-MWC-MultiWii-Flight-Controller-E-/130516118046?pt=LH_DefaultDomain_71&hash=item1e635dea1e

http://www.ebay.com/itm/CRIUS-CO-16-OLED-Display-Module-V1-2-for-MWC-MultiWii-Flight-Controller-E-/130516118046?pt=LH_DefaultDomain_71&hash=item1e635dea1e

Robojuchen, 2015-01-29:
ja tuli vielä mieleen että jos haluaa vielä kehittää kellosta persoonallisempaa niin voisiko olla joku esteettisempi tapa osoittaa ajan kuluneen kuin perus piippaus. Pienen mekaanisen kellon kilautus tms.?

somkra, 2015-02-01:
Tuosta patterin valinnasta, niin onko siinä muuta eroa 3.3v ja 3.7v patterien välillä kuin koko?

Olemme tekemässä molemmat omat munakellot, joten meidän munakellojen kotelloinnit saattavat olla erilaiset. Itse olen koteloinnin jo suunnitellut ja haluan toteuttaa munakellosta mahdollisimman pienen ja siron. 

Miten kaikki johdot ja muut liitetään siihen arduino pro miniin? Tulisiko tehdä jokin piirilevy johon kolvaamalla liitetään johdot vai kolvataanko johdot suoraan arduinon jalkoihin?