matpi/EmbeddedSystemsUTU #16: Spede Spelit reaktiopeli - Anni Ja Eetu

anninen, 2015-01-15:
Teemme Speden Speleistä tutun reaktiopelin arduino nanoa käyttäen. Olemme metallitöissä valmistaneet työlle jalan ja muovityönä teemme kotelon. Löysimme täältä githubista suuntaa antavan koodin peliimme. Olemme hankkineet jo arduinon (funduino) ja joitakin osia meillä saattaa olla jo valmiina. Loput hankimme mahdollisimman pian. Kun olemme saaneet kaikki osat hankittua, valmistamme piirilevyt ja kokoamme työn niin, että voimme lähteä tappelemaan koodin kanssa. - Iivonen ja Kanerva

Robojuchen, 2015-01-15:
Tässä on hvyä projektin aihe! Laitatteko linkkiä siihen koodiin niin katson vähän kannataako sitä käyttää vai olisiko parempi rakentaa oma koodia alusta asti.

Oletteko miettiineet napit jo valmiiksi? Minkälaisella näytöllä tulokset näytetään, onko laitteessa muita efektejä?

Eetuska, 2015-01-23:
Tässä linkki http://petenpaja.blogspot.fi/2013/04/first-project-arduino-spede.html. Mietimme, että olisiko tuossa käytetyn näytön sijasta mahdollista käyttää oled näyttöä? Vai aiheuttaako se liian isoja muutoksia ohjelma koodiin? Hankkisin nappimme partco kaupasta, joten ne ovat nyt valmiina.

Robojuchen, 2015-01-23:
Jos käytätte projektiinne noin valmista konseptia ja koodia siinä pitääkin olla ehdottomasti jotakin muutoksia että tämä olisi oma tuotteenne. OLED tai LCD-näyttö on yksi juttu mikä olisi hyvä lisätä tähän. Pitäisi vielä miettiä miten muuten tätä kehitetään eteenpäin. 

Toki jos haluatte satsata koodipuoleen, teette oman koodin alusta-asti tämä tekee projektista kyllin haastavan...

Eetuska, 2015-01-23:
Voisimme muuttaa koodia niin, että saisimme oled näytön ja haluaisimme myös pelin olevan hieman nopeampi jo heti alussa. Emme kyllä ole tarpeeksi harrastuneita osataksemme tehdä koko koodia itse. Jos jollain tavalla pitää vielä koodia lähteä muuttamaan mietimme äänien tarpeellisuutta. Jokaisesta näppäimestä voisi tulla samanlainen ääni ja jos osaamme, mahdollisen virheen sattuessa tulisi tietty ääni. Voimmeko siis mahdollisesti demoilla käydä näitä asioita läpi? Ja mikä on 74HC595 shift register chip? Mikä sen funktio on työssämme?

Robojuchen, 2015-01-23:
Pientä jalkatyötä on tehtävä teidän itsekin :) käykää malliprojektin koodia läpi huolella kurssilla opitun valossa ja kysykää jos ette jotakin ymmärrä. Demoilla voidaan katsoa lisää sitten. Tuollaisen pelilogiikan ohjelmointi on ihan saavutettavissa sekin mutta muokkaaminen voi olla helpompaa. Tosin koodi jota muokkaa pitää ymmärtää että ihan ilmaista lounasta ei ole luvassa siinä suhteessakaan.

Projektin kytkentäkaaviosta voitte päätellä mihin tuota shiftregisteriä käytetään... Äänien muokkaaminen on hyvä idea, perehtykää miten ääniä käytetään malliprojektissa.

Robojuchen, 2015-01-23:
Vihjeeksi vielä että projektinne koodissa pääohjelmasta (void loop()) käydään aluksi alustusosiossa, sitten peliosiossa kunnes gameover ehto täytyy ja mennään vastaavaan ohjelmaan. Tärkeintä olisi nyt hahmottaa tuo peliohjelman silmukka. Miten on hoidettu näppäimien lukeminen, ledien arpominen ja sytytys/sammutus? Missä kohdassa päivitetään näyttöä ja soitetaan säveliä? Vaikein on ehkä että miten kunkin ledin painamiseen annettu aika, joka ei saa ylittyä on määritely.

Kopioikaa koodi arduino-ohjelmaan ja kommentoikaa siitä ne  kohdat mitä ymmärrätte ja postittakaa minulle sähköpostiin tai tänne githubiin. Huomatkaa että monessa kohdassa käytetään for-loop rakennetta eli jotakin silmukkaa toistetaan laskurin osoittama määrä. lisää tietoa tästä löytyy vaikkapa arduino sivujen referencestä.

Eetuska, 2015-01-23:
Olemmeko oikeassa, jos mielestämme shiftregisteriä ei tarvita ollenkaan jos käytämme olediä? Sen tarkoitus on käytännössä vain säästää arduinon pinnejä? Yritämme käydä koodia vielä huolella läpi, jotta ymmärtäisimme sen. Laitamme sen jälkeen kohdat joita emme tajua.

Robojuchen, 2015-01-23:
hyvä, shiftregister tosiaan ohjaa noita 7-segment näyttöjä joten sitä ette tarvitse jos käytätte muuta näyttöä.

anninen, 2015-04-29:
Meidän käyttämä oled näyttö näyttää vain pikseleitä. Olemme kerran saanut koulun oled näytön toimimaan, joten koodissa ei pitäisi olla vikaa. Tarviiko eri oled näytöt eri kirjaston? Olemme osoitetta koittaneet vaihdella, mutta mikään ei auta. Jussi auta!

Robojuchen, 2015-04-29:
Voisitte kokeilla ylösvetovastuksia vaikkapa 10k molemmista datalinjoista käyttöjännitteeseen. Laittakaapa linkkiä siihen moduulin mikä teillä on.

anninen, 2015-05-04:
http://www.miniinthebox.com/0-96-oled-high-clear-module-board-for-arduino-red_p2229031.html

anninen, 2015-05-04:
Ei auttanut vastukset..

anninen, 2015-05-05:
Hei ni oliko tähän koodin ajamiseen jokin muukin keino kuin tuolla arduino ohjelmalla? Kun koneet eivät enää tunnista meidän arduinoa niin, että voisimme vaihtaa oikean portin ajaaksemme koodimme sisään. Olisimme muuten käytännössä valmiita työmme kanssa, mutta nyt emme saa koodia laitettua laitteeseen.

matpi, 2015-05-06:
Moi... Löysin ohjeen jonka mukaan voisitte kokeilla ohjelmoida Arduinonne... 
http://www.instructables.com/id/Arduino-Examples-2-Use-an-Arduino-as-a-FTDI-Progr/

anninen, 2015-05-17:
Eetun ja mun blogi/raportti löytyy osotteesta spede-projekti.blogspot.com :)

anninen, 2015-05-17:
http://spede-projekti.blogspot.fi/