matpi/EmbeddedSystemsUTU #17: aaaawwwsome uber kattokruunu

rojman, 2015-01-15:
Tarkoituksenamme on yhditää arduino- ja puuntaivutustyö. Puutyönä taivutettuhin kattokruunun sakaroihin tulee kirkkaat led valot. Valojen sytytys ja sammutus toimisisi tietyllä ääniefektillä (taputus ja/tai herätyskellon ään)i. Kirkkaan normaalin valon lisäksi työhön olemme harkinneet myös muita valon väri vaihtoehtoja, kuten punaista tai sinistä. Virta laitteeseen tulisi verkkovirrasta muuntajan avulla.

by. Tea-Maria ja Riku

Robojuchen, 2015-01-15:
Oletteko harkinneet superkirkkaita rgb-ledejä? Niistä saa kirkasta valkoista valoa ja kaikkia muitakin sävyjä. Tuon taputuksen lisäksi olisi hyvä olla jokin muukin tapa ohjata valoja...

Taputuksen sijaan voisi olla vihellysohjaus. Katsokaapa linkki tuosta valopöytä-projektista.

juhasaario, 2015-01-16:
Onneksi puunkautta peilautuva valo ottaa vähän väriä, sillä mikäli on tarkoitus kunnolla valaista huone, on ledien väri ainakin minun silmään aika jäätävä.

MMikko, 2015-01-29:
Tosi hyvä idea! Saanko liittyä ryhmäänne?

rojman, 2015-01-29:
Kirkkaita ledejä olimmekin jo suunnitelleet, sillä tarkoituksenamme on tosiaan valaista kokonainen huone. Ledin pää olisi tarkoitus upottaa jonkinlaisen pleksin sisään, joka voisi muuntaa valoa mukavammaksi. Onko kenelläkään tiedossa millainen muovi/pleksi muuttaisi valoa ehkä hieman lämpimämmäksi, ilman että teho suuremmin kärsisi.
Ajattelimme ryhmässämme, että valo syttyisi taputuksella ja valotehoa voisi säätää vihellyksellä. Onko valotehoa mahdollista säätää, jollain muulla tapaa? Olisiko säätö hyvä tehdä portaittain vai portaattomasti? Portaittain säätäminen olisi varmaankin helpompaa?

P.s. Mikko se on ok.

juhasaario, 2015-01-29:
Noissa superkirkkaissa (>10W) on mainittu se virta ja se on sitten tarkka hommaa ettei ylity yhtään, alituksen ei niin väliä. Ja riittävä jäähdytys takaa pitkän käyttöiän.

Robojuchen, 2015-01-30:
ledeille on olemassa diffusor nimellä himmentimiä/sirouttimia.  valoa voi säättää minkä tahansa sensorin arvojen mukaan. Jos sensori saa portaattomasti arvoja, voi lamppua säätää portaattomasti kun taas digitaalinen signaali kuten taputus tai vihellys säättä helpommin portaitain. Kannattaa valita joko vihellys, tai taputus tai joku muu yksi exoottinen tapa ja sitten joku pomminvarma helppo tapa rinnalle (kytkin tms.)