matpi/EmbeddedSystemsUTU #18: arduino

pekma, 2015-01-16:
Mietimme jatkojohtoa, jossa olisi ääniohjaus eri pistokkeille. Johtoon voisi esimerkiksi yhdistää työ- ja yölampun sekä muita laitteita. Arduino voisi reakoida esimerkiksi taputuksiin ja eri määrillä taputuksia kytkeä ja katkaista virran pistokkeista. Esimerkiksi ensimmäinen pistoke yksi taputus virta päälle-kaksi taputusta virta pois, toinen pistoke kolme taputusta virta päälle-neljä taputusta virta pois. 
Toinen mahdollinen olisi laatikko jossa olisi rivissä kytkimiä. Kun painaa kytkimen päälle, niin arduino painaa sen pois servujen avulla. Arduino painaisi kytkimet pois samassa järjestyksessä, kun mitä itse painaa ne päälle. Laatikolla on vain viihteellinen käyttöarvo.
Joonas ja J-P

juhasaario, 2015-01-16:
Itekkin mietin tota useless-boxia. Se on mukavaa seurattavaa ekan minuutin. Oma lisä olis kapasitanssi- tai induktioanturi tms piilossa ja koppa vois välillä "ennakoida" käyttäjän sormeilua. 

Mikä muuten on koulun kanta verkkojännitteen kanssa puuhasteluun?

pekma, 2015-01-23:
http://m.ebay.com/itm/261652374535?txnId=1558020490016
Tollanen ajateltiin laittaa ohjaamaan verkkovirtaa. Tällä hetkellä mietinnässä millä eri keinoilla voisimme ohjata releitä. Harkinnassa on kauko-ohjaus, ääni ja netti. Nyt olis ne nerokkaat ideat tarpeen ohjaukseen.

pekma, 2015-01-23:
http://www.seeedstudio.com/depot/433mhz-rf-link-kit-p-127.html?cPath=139_140
Meiltä löytyy tämän näköiset lähetin ja vastaanotin, joista ajattelimme lähteä kehittelemään kauko-ohjainta.

Robojuchen, 2015-01-23:
Radio-moduuli on hyvä. Tuo releboardi ei oikein käy kun siinä pitäisi olla tunnistus joka kytkee virran pois aina nollakohdassa. SSR-releet käyvät paremmin verkkovirran kytkemiseen.

Robojuchen, 2015-01-23:
Oletteko harkinneet että käyttäisitte valmiita langattomia pistorasioita, hakkeroisitte vain lähetinmoduulin?

pekma, 2015-01-23:
Radio-moduulin kanssa on ongelmia kantaman kanssa. Saimme ne toimimaan vain niiden ollessa ihan vierekkäin. Onko meillä jokin virhe, vai onko niiden kantama hyvin heikko normaalistikkin?

Robojuchen, 2015-01-23:
pitäisi kantaa 10-30m ongelmitta riippuen paljonko on välissä esteitä. kantamaa voi parantaa juottamalla neljännesaaltojakson mittaisen johdon antenniksi eli 433mhz taajuudella noin 17.2cm. Sekä lähettimeen että vastaanottimeen.

Varmistakaa ettei arduino ole 3.3v asetuksella. Lähettimeen voi kykeä poweriksi korkeamman jännitteen ulkoisesta lähteestä mutta kyllä pitäisi toimia 5v jännitteelläkin.

Robojuchen, 2015-01-23:
Unohtui kysyä: mikä kirjasto? Tarkastakaa johdotkin vielä huolella. Kuulostaa siltä että voisi olla kytkentä virhekin.

pekma, 2015-01-23:
Ton tutoriaalin, minkä linkitin avulla tehtiin. Siinä vastaanotin on laitettu toimimaan 3,3 V.

pekma, 2015-01-23:
Ei toiminut ollenkaan, kun laitoin vastaanottimen viidelle voltille. Pitäisi varmaan etsiä joku toinen ohje jota voisi testata. Vastaanottimessa ja lähettimessä on valmiiksi kolvattu johdot, jotka tosin eivät ole huiman pitkät.

Robojuchen, 2015-01-23:
Ahaa! nyt vasta huomasin tuon teidän tutorialin. Tuo on ihan joku kummallinen ohje. Ottakaa virtualWire kirjasto käyttöön, löytyy mahdollisesti jo koulun koneelta, siinä on transmit ja receive examplet. 5v on suositeltava jännite sitä käytettävissä ja datapinnit tulevat arduinon digitallisiin pinneihin.

Robojuchen, 2015-01-23:
receive pinni on oletuksena muistaakseni 11 ja lähtetinpuolella 12. Kirjasto löytyy täältä jos sitä ei ole koulun koneella valmiina.http://www.airspayce.com/mikem/arduino/VirtualWire/

pekma, 2015-01-23:
Olisiko tähän kytkentäkaaviota? Meilä on ongelmia hahmottaa, että mikä pinni kuuluu tulla mihinkin.

Robojuchen, 2015-01-23:
eikös niissä moduuleissa lue GND ja vcc? ja datalinjat aiemmin sanomani mukaan.http://www.seeedstudio.com/wiki/index.php?title=433Mhz_RF_link_kit

pekma, 2015-01-23:
###Me käytettiin nyt lähettimessä seuraavaa koodia (lähettimen datapinni kytketty arduinon  pinniin 12)

// transmitter.pde
//
// Simple example of how to use VirtualWire to transmit messages
// Implements a simplex (one-way) transmitter with an TX-C1 module
//
// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: transmitter.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
#include <VirtualWire.h>
void setup()
{
Serial.begin(9600); // Debugging only
Serial.println("setup");
// Initialise the IO and ISR
vw_set_ptt_inverted(true); // Required for DR3100
vw_setup(2000); // Bits per sec
}
void loop()
{
const char *msg = "hello";
digitalWrite(13, true); // Flash a light to show transmitting
vw_send((uint8_t *)msg, strlen(msg));
vw_wait_tx(); // Wait until the whole message is gone
digitalWrite(13, false);
delay(200);
}
###Vastaanottimessa käytettiin koodia 

// receiver.pde
//
// Simple example of how to use VirtualWire to receive messages
// Implements a simplex (one-way) receiver with an Rx-B1 module
//
// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: receiver.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
#include <VirtualWire.h>
void setup()
{
Serial.begin(9600); // Debugging only
Serial.println("setup");
// Initialise the IO and ISR
vw_set_ptt_inverted(true); // Required for DR3100
vw_setup(2000); // Bits per sec
vw_rx_start(); // Start the receiver PLL running
}
void loop()
{
uint8_t buf[VW_MAX_MESSAGE_LEN];
uint8_t buflen = VW_MAX_MESSAGE_LEN;
if (vw_get_message(buf, &buflen)) // Non-blocking
{
int i;
digitalWrite(13, true); // Flash a light to show received good message
// Message with a good checksum received, dump it.
Serial.print("Got: ");
for (i = 0; i < buflen; i++)
{
Serial.print(buf[i], HEX);
Serial.print(" ");
}
Serial.println("");
digitalWrite(13, false);
}
}

###Vastaanottimen serial monitoi kirjoittaa jatkuvalla syötöllä seuraavaa tekstiä:
Got: 68 65 6C 6C 6F 
Got: 68 65 6C 6C 6F 
Got: 68 65 6C 6C 6F 
Got: 68 65 6C 6C 6F

###Siis ASCII koodia ja tarkoittaa Hello! Hyvä...  Kantama laitteiden välillä on harmillisesti  maksimissaan n.2 m,... antennin pituus n. 10 cm

pekma, 2015-01-27:
Käsittääksemme jatkojohdon tekeminen on laillista. Releistä jäin miettimään SSR-releitä. Juttelin sähköalan ammattilaisen kanssa ja hänen mielestään SSR-releiden käyttö ei olisi välttämätöntä. Hän on törmännyt niihin vain teollisuudessa suuria tehoisia laitteita ohjattaessa, esimerkiksi piirilevyjenjuottouunissa, jossa tehot ovat toista luokkaa kuin meidän projektissa. Hän oli sitä mieltä, että linkittämiämme relebordeja käytetään yleisesti vaihtovirran ohjaamiseen. Kehoitti kuintenkin välttämään varmuuden vuoksi todella suuritehoisia laitteita. Uskoisin siis, että jos emme rupeaisi silitysrautoja yms. laitteita siihen kytkemään, ei pitäisi olla vaaraa. Normaalien valojen ja pienien elektroniikkalaitteiden ohjaamiseen sen pitäisi hyvin sopia.

Robojuchen, 2015-01-27:
Voitte minun ja Matin puolesta toteuttaa tämän projektin. Tosiaan jos pienempiä kuormia ajetaan eikä esim induktiivisia kuormia kuten moottorit niin pitäisi myös noiden perusreleiden toimia. Ajattelin ensin että jos halutaan ajaa mitä tahansa kodin sähkölaitteita tarvitaan SSR. Mutta osaattehan te tietenkin huolehtia että laitetta käytetään soveltuvasti.Releistä vähän tietoa http://koti.mbnet.fi/erkkak/elektroniikka/releet.pdf

Tutkin itsekin noita laillisuuksia ja totesin että jatkojohdon saa tehdä itse ja koota omaan käyttöön harrastelektroniikkaa. Eiköhän tämä mene noihin osa-alueisiin. Lisäksi velvoitetaan hankkimaan opastus ja ohjeistus asiantuntijalta, minkä olette jo tehneetkin. Eli ei muuta kun antaa palaa :) vaan.

Pidätte huolta, ainakin demoilla, suosittelen muuallakin, että laite ei ole koskaa verkkojännitteessä kun siihen tehdään kytkentöjä. Koteloinnissa ja maadoitusten kanssa on oltava huolellinen. On kokemusta verkkojännitteen maistelusta cd-soittimen korjauspuuhissa ja jäi mieleen että töpsäkkeen voisi vetää pois jatkossa...

Tuossa releboardissahan on vielä optoeristimet että arduinon ei tarvitse olla missän tekemisissä koko releiden elektroniikan kanssa. Voin tuoda teille ensi demoille tuollaisen moduulin testailtavaksi jos haluatte. Itse olen kytkenyt näillä vain DC-virtaa ja käyttänyt suosiolla SSR-releitä aina, niissä on se mukava puoli etteivät napsu kytkettäessä, mutta kokeilllaanpa tällaistakin.

pekma, 2015-02-12:
# Apua kaivataan!

#include <VirtualWire.h>
int rele=4;
void setup()
{
pinMode(rele,OUTPUT);
digitalWrite (rele,HIGH);  
Serial.begin(9600); // Debugging only
Serial.println("setup");
// Initialise the IO and ISR
vw_set_ptt_inverted(true); // Required for DR3100
vw_setup(2000); // Bits per sec
vw_rx_start(); // Start the receiver PLL running
}
void loop()
{
uint8_t buf[VW_MAX_MESSAGE_LEN];
uint8_t buflen = VW_MAX_MESSAGE_LEN;
if (vw_get_message(buf, &buflen)) // Non-blocking
{
int i;
digitalWrite(13, true); // Flash a light to show received good message
// Message with a good checksum received, dump it.
Serial.print("Got: ");
for (i = 0; i < buflen; i++)
{
Serial.print(buf[i], HEX);
Serial.print(" ");
}
Serial.println("");
digitalWrite(13, false);
}
if(
# Miten saamme ohjattua sarjaporttiin tulostuvan viestin avulla relettä? Emme keksi miten sarjaportiin tulevat tiedot on nimetty. Saamme kyllä lähetettyä lähettimellä mitä arvoja haluamme, mutta miten saamme vertailtua haluaamme toimintoa sarjaporttiin tulevaan arvoon?
)
{
  digitalWrite(rele,LOW);
  delay(5000);
}

}

# Tälläista koodia yritimme vastaanottimeen tehdä. Ongelmamme on se, että miten saamme hyödynnettyä vastaanottimeen tulevat viestit.

juhasaario, 2015-02-12:
Jummarsinko oikein, että teillä on ottimessa oikeat datat saatavissa kohdassa...

for (i = 0; i < buflen; i++)
{
Serial.print(buf[i], HEX);
Serial.print(" ");
}

... mutta siiten tie nousee pystyyn?

Robojuchen, 2015-02-12:
tuo kirjasto toimii siten että lähetetty viesti tulee char tyyppisinä (0-255) lukuina ja tallennetaan lukujonoon (array) nimeltä buf [ ]. voitte lukea tuosta jonosta esim. int b = buf[0]; tai b = buf[1]; tms.

sitten vaan sen mukaan mikä luku saadaan...

pekma, 2015-02-24:
Mitä int b = buf[0]; tarkoittaa? Ohjelmaa herjaa, jos sen laittaa ennen void setuppia.Tulee ilmoitus "buf" was not declared in this scope. Mihin kohtaan koodia se pitäisi lisätä? Emme ymmärtäneet miten lukujonoa hyödynnetään niin, että saamme tehtyä jotain. Miten voimme esimerkiksi lähettäessämme toisesta arduinosta luvun 2, saada toisesta arduinossa tapahtumaan toiminnon digitalWrite (rele, HIGH);

pekma, 2015-02-24:
https://github.com/Robojuchen/Mechatronics2013/wiki/Radiomoduulit-433mHz-tai-315mHz
Löydettiin tällainen, joka selvitti aika tavalla asiaa. Tämän avulla onnistuttiin ohjaamaan relettä.

pekma, 2015-02-26:
Vastaanotin ja if -lauseella releen ohjaus:

//
// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: receiver.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
// Tämä versio vastaanottaa lukuja 0-255 ja tallentaa ne array-tyyppiseen muuttujaan,
//eli kokonaislukujen (int) jonoon.
#include <VirtualWire.h>
int rele = 4;

void setup()
{
  pinMode(rele, OUTPUT);
  digitalWrite(rele,HIGH);
  Serial.begin(9600); // Debugging only
    Serial.println("setup");

    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec

    vw_rx_start();       // Start the receiver PLL running
}

void loop()
{
    uint8_t buf[VW_MAX_MESSAGE_LEN];
    uint8_t buflen = VW_MAX_MESSAGE_LEN;

    if (vw_get_message(buf, &buflen)) // Jos viestejä saapuu niin luetaan ne buf-lukujonosta
    {
    int luku =buf[0]; //luetaan muuttujasta niin monta numeroa kun tiedämme lähetetyn
        int luku2= buf[1];// ensimmäisen paikan numero on 0

    // Message with a good checksum received, dump it.
    Serial.println("Got: ");

        Serial.println(luku); // nyt viestin numerot ovat kokonaislukuja joita 
         Serial.println(luku2); //voi käyttää esim. if lauseissa jotka ohjaavat ohjelmaa
        Serial.print(" ");
    }
if (buf[1] == 100)
{
digitalWrite(rele, LOW);

delay (5000);

digitalWrite(rele, HIGH);

delay (2000);
}
   

}

# eli rele alkaa toimimaan viisi sekuntia päälläa ja kaksi sekuntia pois, kun lähetin antaa signaalin. mutta jää pyörimään joko hetkeksi tai pitkäksi aikaa, vaikka lähetin on pois.

Lähettimenkoodi:

// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: transmitter.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
// Lähetetään lukuja väliltä 0-255 ja otetaan ne vastaan kokonaislukuina toisessa päässä
#include <VirtualWire.h>

void setup()
{
    Serial.begin(9600);   // Debugging only
    Serial.println("setup");

    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec
}

void loop()
{
  int luku = 240; // tallennetaan arvot muuttujiin
  int luku2 =100;
     char msg[6] = {(char) luku}; // muunnetaan ne char-muotoon ja tallenetaan lukujonoon
     msg[1] = (char) luku2; //luku meni paikkaan 0 ja luku2 paikkaan 1.

    vw_send((uint8_t *)msg, strlen(msg)); //lähetetään lukujono msg
    vw_wait_tx(); // Wait until the whole message is gone

    delay(200);
}

Robojuchen, 2015-02-26:
Nyt koodi ei ole kaukosäädinmielessä kovin järkevä. Lähetin lähettää samaa viestiä 200 millisekunnin välein ja vastaanottaja laittaa relettä päälle jos lukujonon paikassa 1 on luku sata. Huomatkaa että siellä paikassa on se sama sata niin kauan kuin sinne tallenneta jotakin muuta. Eli joko nollaatte lukujonon paikat aina sopivassa paikassa tai sitten asetatte nuo toimintokäskyt siihen ohjelmalohkoon mikä suoritetaan vain jos uusi viesti on saapunut, eli tuo if(vwmessage(buf jne...) tarkistaa onko viestejä tullut ja sitten viestin saavuttua tekee if lausetta seuraavien lohkosulkeiden sisäisen koodin.

Voisitte kehittää tuohon lähettimeen että se lähettää viestin vain jos nappia painetaan...

pekma, 2015-03-17:
## Lähettimen koodi

// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: transmitter.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
// Lähetetään lukuja väliltä 0-255 ja otetaan ne vastaan kokonaislukuina toisessa päässä
#include <VirtualWire.h>

int n1=2;
int n2=3;
int n3=4;
int n4=5;
char msg[6];
int laskuri = 0;

void setup()
{
    Serial.begin(9600);   // Debugging only
    Serial.println("setup");

    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec
    
     for(int i = 0; i <4; i++)
   {
    msg[i] = 0; 
   }
   
  pinMode(n1,INPUT);

  pinMode(n2,INPUT);

  pinMode(n3,INPUT);
   
  pinMode(n4,INPUT);
}

void loop()
{
  
  for(int i = 0; i <4; i++)
 {
   msg[i] = digitalRead(i+2); 
  //laskuri = laskuri + msg[i];
 }
  laskuri = (msg[0]+msg[1]+msg[2]+msg[3]);
 
 if (laskuri > 0)
 {
    vw_send((uint8_t *)msg, strlen(msg)); //lähetetään lukujono msg
    vw_wait_tx(); // Wait until the whole message is gone
    
 }
  
  for(int i = 0; i <4; i++)
 {
  msg[i] = 0; 
 }
  laskuri = 0;
  

}

## Vastaanottimen koodi

//
// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: receiver.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
// Tämä versio vastaanottaa lukuja 0-255 ja tallentaa ne array-tyyppiseen muuttujaan,
//eli kokonaislukujen (int) jonoon.
#include <VirtualWire.h>

void setup()
{
    Serial.begin(9600); // Debugging only
    Serial.println("setup");

    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec

    vw_rx_start();       // Start the receiver PLL running
}

void loop()
{
    uint8_t buf[VW_MAX_MESSAGE_LEN];
    uint8_t buflen = VW_MAX_MESSAGE_LEN;

    if (vw_get_message(buf, &buflen)) // Jos viestejä saapuu niin luetaan ne buf-lukujonosta
    {
    int luku =buf[0]; //luetaan muuttujasta niin monta numeroa kun tiedämme lähetetyn
        int luku2= buf[1];// ensimmäisen paikan numero on 0
        int luku3=buf[2];
        int luku4=buf[3];
        

    // Message with a good checksum received, dump it.
    Serial.println("Got: ");

for(int i=0; i<4; i++)
{
        Serial.println(buf[i]); // nyt viestin numerot ovat kokonaislukuja joita 
         
}
    }

}

Meillä on ongelmana se, että emme saa vastaanotettua viestiä oikein, eikä kolme neljästä napista näytä tekevän mitään. Vastaanottimeen tulee:

Got:
1
80
0
0

Missä on vika?

pekma, 2015-04-07:
Emme saaneet ylläolevaa koodia yrityksistämme huolimatta toimimaan. Tähän koodiin olimme tehneet vuokaaviot. Osaatko kertoa, miksi ylhäällä oleva koodi ei toimi?

Pähkäillessämme miten etenemme, päädyimme kirjoittamaan koko koodin eri tavalla ja näin kauko-ohjaimen ja releen toimimaan. Kuitenkin välillä rele jää vilkkumaan päälle-pois asentoon ilmeisesti johtuen  liian nopeista painalluksista johtuen. Tämä kuitenkin korjaantuu nappia uudelleen painamalla, mutta tämä epäkohta pitäisi vielä koodissa korjata. Tähän voisimme kaivata apua. Tähän koodiin emme ole vielä saaneet tehtyä vuokaavioita johtuen siitä, että koko koodi meni uusiksi. Alla lähettimen ja vastaanottimen koodit. Koodit on kirjoitettu kiireellä, emmekä ole oikein ehtineet kirjoittaa selityksiä. Tarkoitus olisi kuitenkin tämä lähiaikoina tehdä.

## Lähetin

// See VirtualWire.h for detailed API docs
// Author: Mike McCauley (mikem@airspayce.com)
// Copyright (C) 2008 Mike McCauley
// $Id: transmitter.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
// Lähetetään lukuja väliltä 0-255 ja otetaan ne vastaan kokonaislukuina toisessa päässä
#include <VirtualWire.h>

  int luku = 1;
  int luku2 = 1;
  int luku3 = 1;
  int luku4 = 1;

void setup()
{
    Serial.begin(9600);   // Debugging only
    Serial.println("setup");

    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec
pinMode(2, INPUT);
pinMode(3, INPUT);
pinMode(4, INPUT);
pinMode(5, INPUT);
}

void loop()
{
 

 if (digitalRead(2)==HIGH)
{
  luku = 2;
  luku2 = 1;
  luku3 = 1;
  luku4 = 1;
   char msg[6] = {(char) luku}; // muunnetaan ne char-muotoon ja tallenetaan lukujonoon
     msg[1] = (char) luku2; //luku meni paikkaan 0 ja luku2 paikkaan 1.
msg[2] = (char) luku3;
msg[3] = (char) luku4;

    vw_send((uint8_t *)msg, strlen(msg)); //lähetetään lukujono msg
    vw_wait_tx(); // Wait until the whole message is gone

    delay(200);
   
    {
      luku = 4;
      luku2 = 4;
      luku3 = 4;
      luku4 = 4;
  
 char msg[6] = {(char) luku}; // muunnetaan ne char-muotoon ja tallenetaan lukujonoon
     msg[1] = (char) luku2; //luku meni paikkaan 0 ja luku2 paikkaan 1.
msg[2] = (char) luku3;
msg[3] = (char) luku4;

    vw_send((uint8_t *)msg, strlen(msg)); //lähetetään lukujono msg
    vw_wait_tx(); // Wait until the whole message is gone
    delay(200);
   
 }  
  }
 
[... 248 lines cut ...]
    vw_wait_tx(); // Wait until the whole message is gone

    delay(200);
    {
      luku = 4;
      luku2 = 4;
      luku3 = 4;
      luku4 = 4;
  
 char msg[6] = {(char) luku}; // muunnetaan ne char-muotoon ja tallenetaan lukujonoon
     msg[1] = (char) luku2; //luku meni paikkaan 0 ja luku2 paikkaan 1.
msg[2] = (char) luku3;
msg[3] = (char) luku4;

    vw_send((uint8_t *)msg, strlen(msg)); //lähetetään lukujono msg
    vw_wait_tx(); // Wait until the whole message is gone
    delay(200);
   
 }  
  }
 
 
  }  

## include kohtaan pitää lisätä molemmissa koodeissa < VirtualWire.h > ilman välejä

Robojuchen, 2015-04-08:
Hyvää työtä olette tehneet kun saitte tehtyä koodin toimimaan. Tuo aiempi koodi on vain aika paljon tehokkaampi ja elegantimipi. Keksin nyt mikä siinä oli ongelmana. Tuo virtual Wire toimiikin siten että sille annetaan vaan muistialuuen alkuosoite ja montako tavua se lähtettää siitä eteenpäin lukien. Vastaavasti vastaanottimelle ilmoitetaan montako tavua (byte) se vastaanottaa ja mihin se ne tallentaa. Ts. Tällä voi lähtettää ihan suoraan int-lukujonon ja tallentaa toisessa päässä kunhan ilmoittaa oikean määrän tavuja minkä tallennus ja lähetys vaatii. Alla toimivat esimerkit:

Lähetin:
    #include <VirtualWire.h>

    // See VirtualWire.h for detailed API docs
     // Author: Mike McCauley (mikem@airspayce.com)
     // Copyright (C) 2008 Mike McCauley
     // $Id: transmitter.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
     
    
        
    int n1=2;
     int n2=3;
     int n3=4;
     int n4=5;
     int msg[4];
     byte viestinPituus = 8; // neljä int-lukua on 8-tavua(byte)
     int laskuri = 0;
    
    void setup()
     {
     Serial.begin(9600); // Debugging only
     Serial.println("setup");
    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec
    
     for(int i = 0; i <4; i++)
    
    
    {
     msg[i] = 0; 
     }
    
    pinMode(n1,INPUT);
    
    pinMode(n2,INPUT);
    
    pinMode(n3,INPUT);
    
    pinMode(n4,INPUT);
     }
    
    void loop()
     {

    for(int i = 0; i <4; i++)
     {
     msg[i] = digitalRead(i+2); 
     //laskuri = laskuri + msg[i];
     }
     laskuri = (msg[0]+msg[1]+msg[2]+msg[3]);
    
    if (laskuri >0)
     {
     vw_send((byte*)msg, viestinPituus); //lähetetään lukujono msg
     vw_wait_tx(); // Wait until the whole message is gone
    
    }

    for(int i = 0; i <4; i++)
     {
     msg[i] = 0; 
     }
     laskuri = 0;
    
    }

Robojuchen, 2015-04-08:
vastaanotin:

    #include <VirtualWire.h>
    //
     // See VirtualWire.h for detailed API docs
     // Author: Mike McCauley (mikem@airspayce.com)
     // Copyright (C) 2008 Mike McCauley
     // $Id: receiver.pde,v 1.3 2009/03/30 00:07:24 mikem Exp $
    
    
    byte viestinPituus = 8; //neljä int-lukua vaatii 8 byteä tilaa
       void setup()
     {
     Serial.begin(9600); // Debugging only
     Serial.println("setup");
    // Initialise the IO and ISR
    vw_set_ptt_inverted(true); // Required for DR3100
    vw_setup(2000);  // Bits per sec
    
    vw_rx_start();       // Start the receiver PLL running
    
    }
    
    void loop()
     {
       int data[4]; // tähän tallennetaan tuleva data. Luodaan tässä että se olisi tyhjä joka kerta.
    
    if (vw_get_message((byte*)data, &viestinPituus)) // Jos viestejä saapuu niin luetaan ne buf-                 lukujonosta 
    {
    
    // Message with a good checksum received, dump it.
    Serial.println("Got: ");
    
    
    for(int i=0; i<4; i++)
     {
     Serial.println(data[i]); // nyt viestin numerot ovat kokonaislukuja joita voi lukea lukujonosta
    // int luku0 = data[0]; jne
    }
     }
    
    }

pekma, 2015-05-08:
Projekti valmis!
Lisää juttua löytyy blogista

http://arduinoaopiskelemassa.blogspot.fi/