matpi/EmbeddedSystemsUTU #19: Patrik ja Henri, Langaton lämpömittari LCD näytöllä

PatrikkiJaHenrikki, 2015-01-16:
Lämpötila-anturi on kiinni radiolähetinpiirissä, vastaanottopiiriltä data ohjataan arduinon kautta LCD näytölle. Jotain tälläsen tapasta.
Arduino lisäksi käyetään lämpötila-anturia ja sen kytkemiseen tarvittat komponenti, langattomaan tiedonsiirtoon radioliikenteen lähetys-ja vastaanottopiiri ja LCD näyttö.
Vielä paljon mietittävää miten homman toteuttaa, mutta kaippa sitä selviiää.

Robojuchen, 2015-01-16:
Hyvä tällainen sääasema on mukava projekti ja hyödyllinenkin. Tässä voi satsata eri asioihin. Suosittelen dht22 yhdistettyä kosteus ja lämpösensoria. Myös ilmanpainetta on helppo mitata. Ilmanlaatukin voisi olla kiva.

Tarvitsette 2 kpl arduino pro miniä ja vaikkapa 433mhz radiomoduulit. Pääsette kokeilemaan jotakin näistä heti seuraavalla demolla!

PatrikkiJaHenrikki, 2015-01-17:
Mitäs nää maksaa ? Arduino uno kyllä jo löytyy, että ei tarvittas ku yks tommone pro mini.

PatrikkiJaHenrikki, 2015-03-27:
Onko mahdollista saada kirjasto hankkimaamme Oled-näyttöön, jos sinulta kirjasto löytyy?
Viime demolla keskustelimme asiasta.

Robojuchen, 2015-03-30:
https://learn.adafruit.com/monochrome-oled-breakouts/overview

tuossa tutorialissa on kaikki tieto kun sitä menette sivu kerrallaan eteenpäin. Siinä on linkki myös kirjastoon. Kiraston github repon oikeasta reunasta painatte download zip ja sitten asennatte kirjaston koneelle.