matpi/EmbeddedSystemsUTU #20: 2D piirturi, ei plotteri

juhasaario, 2015-01-16:
A4 paperille (tod näk traktorille) magee X-Y piirturi. Inputti käyttäjän kädenliikkeistä ilmassa, (tod näk kapasitanssi), vähän softausta ja bob's your uncle.

Rakenne halvin mahdollinen eli vaneri. Kelkkojen liike kierretangolla, motit näihin vanhoista akkukoneista ja noparit samaten. 

Kovasti tekisi manuaalijyrsintä mieli osien tarkkuuden takia, mutta kun ei ole niin ei ole. Noo, tehdään pieni välys, no bigi.

Robojuchen, 2015-01-16:
Mielenkiintoinen projekti! Akkuporan moottoreilla voi jäädä tarkkuus vähän heikoksi,sikäli askelmoottorit tai jotkin hitaat vaihteistomoottorit olisivat paremmat. Askelmoottoreissa ei tarvitsisi enkoodereita tarkkailemaan liikettä, toisin kuin vaihteistomoottoreissa.

Plotterin saisi helposti ajamaan g-koodia, arduinoa on käytetty tähänkin. Toki tuo idea käsiohjauksesta on oikein intressantti

juhasaario, 2015-01-16:
Tnx mään. Intresantti tässä on "se" juttu, ja häärreguud, mitä oikein ajattelinkaan!!? Jätinhän mainitsematta vallan sijainnin tarkkailun kierretangoista optohaarukoilla (vanhasta rotasta ilmaiseksi tietenkin) 

Ja motit tulee suoraan kierretankoihin ilman välitystä, grrr, tarkoitus on mennä täysiä! Ja mitä tarkkuuteen tulee, arvelemnpa ardun suoriutuvan tehtävästään hyvin. Kelkat mahd kevyiksi inertian minimoiseksi ja silleen.
Kierretankojen hallitsematon värinä saattaa olla suurin nopeutta rajoittava tekijä. We'll see.

Ja hiukan lisää nippeliä. Ennen jokaista ajoa, vedetään molemmat suunnat katkaisijoita vasten "tappiin" ja kalibroidaan. Vähän ehkä turhaa, mutta mikäli mennään laskuissa sekaisin tulee toiminnan mahdollistava savu ulos.

Plotterin saisi kyllä, mutta se on jo ihan toinen juttu. Tämä kun on semmoinen "taidetta" tekevä kone, vetelee nääs viivaa pitkin paperia ristiin rastiin. AI niin, ja "kynän" painamismomenttia säätelee kolmas motti. Kynänä 6 mm puristehiili.

Robojuchen, 2015-01-16:
Vauhtia kyllä piisaa jos akkuporan koneilla mennään, mitenköhän tosiaan kierretangot reagoi 550 rpm menoon. Miten ajattelit säädellä moottorien suuntaa ja nopeutta? Vaatii aika järeät h-sillat. Minulla olisi suositus yhdestä halvasta ja hyvästä mallista. Laittelen siitä linkkiä myöhemmin.

Tuli vielä mieleen että jos purkaisi parista vanhasta printteristä johteet ja hihnat sekä moottorit pärjäisi vähän halvemmilla ajureilla.

 Tästä tulee varmasti viihdyttävä ja vinkeä vehje

juhasaario, 2015-01-17:
Nopeuden säätelyyn ajattelin koneiden omia nopareita käytellä, suunnanvaihto vielä hieman auki, mutta jos muu ei onnistu, niin releellä tahi feteillä korvaan sormivalitsimen. Anyways, en ole koskaan puukottanu akkukoneiden noparia muuten kuin puhdistustarkoituksissa, mutta eiköhän sieltä ole normi pwm kaivettavissa, tai jopa suoraan ardulla käskettävissä?

Noi myllythän vetää sikana juicyä käynnistyäkseen, ilman kuormaakin about 5A ja siitä ylös. Poweriksi vanha pc:n virtalähde, jos jaksaa pukata tarpeeksi. Kattoo sitä sitten, voi olla että pieneksi jää.

Mutta pistä sitä linkkiä vaan, aina kiinostaa romppeet.

Robojuchen, 2015-01-26:
http://www.ebay.com/itm/Double-BTS7960B-DC-43A-Stepper-Motor-Driver-H-Bridge-PWM-For-Arduino-smart-Car-/200950936475?pt=LH_DefaultDomain_0&hash=item2ec99bf39b

Tässä on halvin isoa virtaa kestävä h-silta. Yksi moduuli on siis vain yhdelle moottorille, nämä ovat minulla toimineet hyvin isomassakin robotissa akkuporan moottoreilla. NImittäin yritin kerran noita nopareita availla mutta en oikein saanut selvää miten ne toimi ja datalehtiä ei aina löydy. Lisäksi suunnanvaihto on tosiaan vähän kinkkistä kun se akkukoneessa toteutetaan tosiaan käyttäjän nakkisormella. Jos et jostain löydä halvalla isoa virtaa kestäviä releitä niin halvempaa ja helpomaa on ostaa valmiit H-sillat kuin rakentaa irto-feteistä. Hattu kyllä nousee korkealle (ja arvosana) jos suunnittelet vielä oman H-sillan tähän  :)

juhasaario, 2015-01-26:
50 ämpäriä, ei paha.

Robojuchen, 2015-01-29:
Tässä kaksi h-siltaa yhdessä shieldissä ja voi siis laittaa arduino UNO:n selkään. Luulisin että tämä jaksaa pyöritellä noita akkuporan motteja jos niissä ei ole hirveästi kuormaa. Hinta on ainakin käsittämättömän halpa tuon tehoiselle setille. Tilasin yhden itselle koekäyttöön.

http://www.ebay.com/itm/310780916584

juhasaario, 2015-01-29:
Rc-noparia ei ole vielä mainittu.

Robojuchen, 2015-01-29:
Jaa, niistä ei ole kokoemusta kuin harjattomille moottoreille tarkoitetusta, mikä taas ei taida tähän sopia.