matpi/EmbeddedSystemsUTU #21: metallinpaljastin

Jannola, 2015-01-16:
Suunnitelmissa olisi valmistaa metallinpaljastin. Löydettiin netistä valmis koodi, mutta tarvittavat osat jäi vähän epäselväksi. Pitäisi hieman pohtia lisää miten koodia voisi muuttaa tai voisiko siihen lisätä jotain uutta.

Jannola, 2015-01-16:
http://blog.atmel.com/2013/07/25/build-your-own-metal-detector-with-an-arduino/ tuolta sivulta löyty se koodi

Janne Annola ja Teemu Aaltonen

Robojuchen, 2015-01-26:
Tämä on mielenkiintoinen laite kyllä. Mietittävää on vain että miten tekisitte tästä omannäköisenne projektin. Mitä osa-aluetta voisitte kehittää ettei projekti olisi silkkaa kopiointia. 

Elektroniikan kehittäminen etsimen osalta on aika exoottista elektoriniikkaa, en ehkä suosittele ellette ole harvinaisen vihkiytyneitä asiaan. Koodi on aika kehittynyttä syvemmän "rauta" tason ohjelmointia. Lähinnä voisitte kehittää tuota käyttöliittymää. Kehittäisitte laitteeseen näytön joka osoittaa graafisesti signaalin voimakkuuden ja jotakin muutakin hauskaa/hyödyllistä. Lisäksi arvosanaa voi korottaa laitteen viimeistelyllä, virransäästöominaisuuksilla, säätömahdollsiuuksilla yms. 

Sinällään hyvä projekti että tästä voi saada oikean hienon tuotteen!

Jannola, 2015-01-29:
Kuullostaa hieman hankalalta. Innostuttii viime luennolla siitä RFID hommasta, niin lukeeko se homma sitä lätkää metallin läpi? Ku mietittiin että ois hyvä tehdä asekaappi ja sille olisi tarvetta. Ilmeisesti se vois olla hieman helpompi tehdä, kun ei olla kovin vihkiytyneitä tähän hommaan. 

Jos se ei lue metallin läpi, niin on vissii mahdollista kytkeä myös kaapin ulkopuolle se luku juttu?

Robojuchen, 2015-01-29:
RFID tunnistimen voi viedä ulos kaapista, metallin läpi ei luultavimmin pysty lukemaan. Tämä olisi hyvä projekti.

Ei tuo metallinpaljastinkaan ole vaikea, siinähän on valmis pohja johon vaan lisätään vähän juttuja joita olette jo demoilla tehneetkin. Kumpi vain on hyvä projekti.

Jannola, 2015-01-29:
Tuo asekaappi kiinnostais eniten tarpeellisuutensa vuoksi. Mietittiin semmosta että RFID lukulaite olisi kaapin sisässä ja kaapin päällä olisi reikä/muovisuoja josta voisi aukaista oven lätkällä. Olisi mukava jos kaappiin saisi varashälyttimen joka laukeaisi rynkytyksestä ja alkaisi huutamaan. Onko tällaisia sensoreita olemassa? Onko sulla tietoa sähkölukoista, jotka soveltuisi tähän työhön. Ajatuksena olisi että ovi aukeaa itsestään kun lätkän asettaa lukulaitteelle. Hienosäätönä vois olla, että kaappi päästää pienen äänen kun sen avaa ja lukitsee.

Jannola, 2015-01-29:
Olisiko kiihtyvyysanturi soveltuva tähän käyttöön?

Robojuchen, 2015-01-30:
Moottori, askelmoottori tai servo voi vääntää lukkoa. Samoin lukituksen voi hoitaa solenoidilla. Noissa on eri huonoja ja hyviä puolia. Rynkytyksen tunnistimena voi käyttää kiihtyvyyssensoria tai tärinäsensoria.

Jannola, 2015-02-25:
Hei!

Olen tilaamassa osia ja haluaisin varmistuksen, että ovatko osat oikeita.

http://www.ebay.com/itm/Security-RFID-Proximity-Entry-Door-Lock-Access-Control-System-500-User-10-Keys-/400547349442?pt=LH_DefaultDomain_0&hash=item5d427b83c2

Saako tuolla tehtyä sillain, että käyttää salasanaa sekä rfid lätkää oven aukasemiseksi?

Sitte oli ajatuksena oli lukitukseen, että ovessa olisi salpa joka "makaa" ovessa olevan hahlon päällä  ja kun ovi aukeaa solenoidi vetää esim rautalangan avulla salvan toisesta päästä. Jolloin salpa nousee ja ovi pääsee aukeamaan. Jos kysessää on vetosolenoidi niin miten sitten solenoidin metallitapin palautiminen takaisin onnistuu? 

http://www.spelektroniikka.fi/tuotteet/solenoidit-sahkomagneetit-moottorit-solenoidit/vetosolenoidi-24v-g-1250-103550

Kun ovi ei ole enään lukossa niin haluaisin että toinen solenoidi työntäisi oven raolleen jolloin tarkoitus olisi käyttää tätä: 

http://www.spelektroniikka.fi/tuotteet/solenoidit-sahkomagneetit-moottorit-solenoidit/shindengen-pushpull-solenoidi-144c-series-103549

Sitte tommonen vois olla hälytysäänenä, jos joku koittaa aukasta ovea ennen kuin ovikoodia on laitettu. http://www.ebay.com/itm/Piezo-Electronic-Tone-Buzzer-Alarm-95DB-Continuous-Sound-12V-Mounting-Hole-EVHS-/161610981279?pt=LH_DefaultDomain_0&hash=item25a0c3df9f

Sireenin laukaisemiseen rynkyttäessä: http://www.ebay.com/itm/GY-521-6-DOF-MPU-6050-Module-3-Axis-Accelerometer-Gyroscope-Module-for-Arduino-E-/361194940459?pt=LH_DefaultDomain_0&hash=item5418e5682b

Virtana vois olla joku mopon akku ja tietty näiden lisäksi arduino. 

Tuleeko sulla mielee mitä vois vielä tilata ja kanttisko tehä jotai toisin? Kiitos!

Jannola, 2015-03-03:
```javascript
function fancyAlert(arg) {
  if(arg) {
    $.facebox({div:'#foo'})
  }
}
```

juhasaario, 2015-03-05:
Hyvä että kuoppasitte tuon kyseisen metallinetsimen, ite en saanu sitä havaitsemaan juuri mitään yli kymmenen sentin syvyydestä vaikka kuinka nostelin kelan jännitteitä ja piuhan kierroksia.

Robojuchen, 2015-03-05:
Huomasin että en ole tätä viestiä huomannutkaan aiemmin.

Kannattaa valita ( jos laite ei tule verkkovirtaan) pienivirtaisempi RFID moduuli. Tuo teidän malli vie nopeasti akun tyhjiin. 

Nämä moduulit ovat pienivirtaisia ja halpoja kuin leipä:

http://www.ebay.com/itm/MFRC-522-RC522-RFID-Reader-IC-Card-Proximity-Module-for-Arduino-/111406616819?pt=LH_DefaultDomain_0&hash=item19f05a10f3

Tuo on hyvä idea että solenoidi avaa lukon ja nopeasti työntää sen auki niin ei kulu turhaan virtaa solenoidin huudattamiseen. Solenoidi palaa takaisin itsestään kun virta katkaistaan. Katsokaa että lukon aukaisin on pull mallia ja oven aukityöntäjä push mallia.

kiihtyvyysanturiksi suosittelen adxl345. pienivirtainen ja siinä on valmis rynkytyksen tunnistus moodi.