matpi/EmbeddedSystemsUTU #22: Pehmolelu, liikkeentunnistimella

hmsirk, 2015-01-17:
Ajatuksena olisi tehdä itse tai käyttää hyödyksi valmista pientä pehmolelua, jonka silmien tilalle tulisi liikkeentunnistin. Lelua lähestyessä se liikuttaisi sitten esim. toista tassua tms.

Robojuchen, 2015-01-22:
Hyvä projektin aihe! Harrastelijapiireissä on ollut suosittuja "seuralais robotiti"  Tähän voisi miettiä jotakin erikoisempaakin vuorovaikutusta. Reagoi kun twitteriiin tulee viesti tms. Tai sitten vois miettiä että nalle ääntelee jotekin... kehittäkää vielä ideaa vähän. 

Toisaalta voisiko lelusta saada jollakin tavalla hyödyllisen...vaihtoehtoja on paljon

hmsirk, 2015-01-24:
Joo, Koitan miettiä vielä vaihtoehtoja, tosin kun teen tämän työn yksin, niin ajattelin ettei siitä tulisi liian monimutkainen...
T: Heidi S

Robojuchen, 2015-01-26:
Ei tarvitse liian monimutkaista tehdä. Miniminä olisi että projektissa käytetään kolmea erilaista sensoria/aktuaattoria/näyttöä/valoja tms. elementtiä. Olisiko liiketunnistinta parempi jokin sensori joka mittaa vähän etäisyyttäkin? 

Lisäksi voisi olla vaikkapa ääntelyä, tai nalle voi hakea tietoa netistä tai nallesta voi hakea jotakin sensorilukemia netin yli. Nämä wifi ja ethernet moduulit ovat kovin halpoja eivätkä mienkään vaikeita käyttää.

Voisko nalle ragoida kosketukseen lämpösensorilla? Väreihin? :)

Tassunheilutukseen suosittelen servoja jos pieni surina ei haittaa. Vaihteistomoottorit tai askelmoottorit ovat sitten hiljaisempia mutta eivät niin vahvoja.

hmsirk, 2015-01-29:
Mietin sellaista, että voisi olla tosiaan joku sensori, mikä mittaisi etäisyyttä. Ja sitten tietyllä etäisyydellä voisi servon avulla tassu heilua ja kun lähemmäs vielä tulee, niin voisko ajatella, että nallen rinnassa olisi oled-näyttö esim, mihin syttyisi esim. sydängrafiikka. Onkohan liian monimutkainen meikäläiselle?:)

hmsirk, 2015-01-29:
Olisiko tarvikelistaan nämä: Infrapunasensori, Arduino uno, Oled-näyttö?

Robojuchen, 2015-01-30:
Suosittelen arduino pro miniä että saadaan mahtumaan nallen sisään paremmin ja ettei nallen patterit lopu niin nopeasti. Paristopuolta voisit miettiä myös. Tuosta oled näytöstä tulee varmasti kiva efekti. Servoja saisi olla pari kolme kappaletta siltä varalta että joku on secundakappale ja saa monipuolisempia liikkeitä nallelle esim. kahdella servolla. Alkaa kuulostaa tämä paketti aika hyvältä eikä ole varmasti liian hankalakaan setti.