matpi/EmbeddedSystemsUTU #23: Joulukalenteri Juho & Sara

Mikne, 2015-01-17:
Joulukalenteri ois tarkotus tehdä. Luukku / luukut ois tarkoitus avautua joko automaattisesti 24h välein tai luukut voi saada auki tietyssä aikavälissä.

sipara, 2015-01-20:
Rupesimme epäilemään omaa taitotasoa ja ajanriittävyyttä tämän projektin kanssa. Löysimme opinsys:in nettisivulta tietoa ohjelmoitavasta nallesta. Nallessa on esimerkiksi liiketunnistimia ja se heiluttaa käsiä ihmisen kulkiessa ohi. Sivu josta löysimme enemmän tietoa: https://github.com/opinsys/ohjelmointi-opetuksessa

Mikne, 2015-01-23:
int analoginput = A0; // Potentiometrin keskimmäinen jalka kytketään pinniin a0

int arvo = 0; // jännitteen arvo tallentaan arvo -nimiseen muuttujaan 

#include <Servo.h>

Servo ekaServo; // luodaan servo-objekti nimeltä ekaServo

void setup()

{

ekaServo.attach(9); //käynnistää ekaServo nimisen servo-objektin pinnissä 9

ekaServo.write(77); // asettaa servon keskelle, 90 asteen kulmaan

delay(1000); // servolle pitää antaa aikaa liikkua

  
// Seuraava komento avaa sarjaliikenteen tietokoneelle nopeudella 9600 bps:

Serial.begin(9600);

}

void loop()

{

// Sitten luetaan jännite analogisesta inputista:

arvo = analogRead(analoginput);

// Tulostetaan saalis sarjaporttiin:

Serial.print("Jännite = " );

Serial.println(arvo); // "Serial.println" komento vaihtaa riviä tulostuksen jälkeen

// pieni tauko:

delay(1);

if (arvo > 400)

ekaServo.write(0); // käskee servon toiseen äärilaitaan, 0-asteen kulmaan

delay(100);

ekaServo.write(77); // käskee servon vastakkaiseen äärilaitaan, 180 asteen kulmaan

delay(100);

}

Mikne, 2015-01-23:
Kokeiltiin tällästä koodia, jota vois käyttää nalleen käden heilutukseen. Käytettiin infrapunasensoria ja pientä servoa testauksessa. Jos DELAY:t on pidempiä, se vaikuttaa myös siihen kuinka tiheästi serial monitoriin tulostuu etäisyys. Jos delayt on pitkiä, myös servo reakoi hitaasti eikä riitä pelkkä heilautus, vaan kättä pitää pitää tavallaan tauon ajan verran tarpeeksi lähellä sensori, jotta servo reakoi tähän. Haluttaisiin tietää, miten tulostus on nopea ja reakointi nopea, mutta niin ettei servon liikkeen tarvitse olla tosi nopea tai lyhyillä tauoilla.  Voiko loopiin tehä uutta aliohjelmaa/tms, jotta tauot eivät vaikuttaisi tulostuksen nopeuteen/hitauteen?

Robojuchen, 2015-01-23:
delayn pystyy totetuttamaan ilman että ohjelman muu suoritus pysähtyy. tähän on olemassa esimerkki arduino-ohjelmassa, "blink without delay". Toisaalta servoa pitää liikuttaa servo kirjastosta löytyvän "sweep" esimerkin mukaisesti for-loop silmukalla jos haluaa että se liikkuu hitaasti ja sulavasti. Tutkikaapa aluksi vaikka sitä servo esimerkkiä.

Robojuchen, 2015-01-26:
Tähän nalleprojektiin pitäisi vielä saada jotakin persoonallista ja erityistä vaikkapa tuhon esimerkkiprojektiin verrattuna. Voisi nalle aistia ympäristöään jollakin sensorilla? Tai puhua/äännellä? Hakea tai antaa tietoa internetistä? Ideoikaa vähän lisää. Yksi sensori ja servot on vähän turhan yksinkertainen koko projektiksi, ellei sitten ala ihan opettamaan nallea kävelemään :)

jktmik, 2015-01-29:
Siihen tulee ledisilmät, kaulapanta jossa näyttö, jne... jotka alkaa välkkymään kun heiluttaa nallelle kättä!

jktmik, 2015-01-29:
Saimme koodattua infrapunasensorin kahteen lediin.. Tarkoitus olisi että ledit syttyvät esim. 50cm etäisyydeltä 

Tähän vielä kun saisi esimerkiksi servot ja näytön yhdistettyä.

Robojuchen, 2015-01-30:
Hyvältä vaikuttaa! Esim kahdella tai kolmella servolla  saa jo kivan monipuolisia ja elävän näköisiä liikeitä. Valitsette vain minkälainen näyttö olisi kiva.

Robojuchen, 2015-01-30:
Servoja on muuten hyvinkin pieniä niin arduino pro miniä käyttämällä voitte tehdä hyvin pienestäkin nallesta projektin jos sellainen kiinnostaa.

sipara, 2015-02-27:
Mietittiin nallen kokoa ja se voisi olla just tosi pieni ja tehdään servojen kanssa eri toimintamalleja. Laitan tähän osia, joita ainakin tarvitaan. Jos kommentoit tätä listaa. Katsoin ebaysta näitä osia ja siellä tulee aina pitkä lista vähän erihintaisia, mutta saman näköisiä laitteita. Mikä lieneen kaikissa eroina sitten.

LISTAA:
- Details about  Module Atmega328 5v 16M+USB2.0 To TTL 6Pin CH340G Converter F Arduino Pro Mini – oisko hyvä paketti?
 http://www.ebay.com/itm/Module-Atmega328-5v-16M-USB2-0-To-TTL-6Pin-CH340G-Converter-F-Arduino-Pro-Mini-/191146731721?pt=LH_DefaultDomain_0&hash=item2c813ba8c9

- CRIUS CO-16
http://www.ebay.com/itm/NEW-CRIUS-CO-16-OLED-Display-Module-V1-2-for-MWC-MultiWii-Flight-Controller-S-/391067365471?pt=LH_DefaultDomain_0&hash=item5b0d6e705f

mitä tarkoittaa controller B tai controller S, onko jotain näytön kokoja vai mitä ?

- Pieniä servoja, esim jotain tän kokosia, mitä löysin kouluelektroniikasta x 4-6
http://www.kouluelektroniikka.fi/cgi-bin/shop.cgi?action=prodshow&usr=79ih1ouvfpp2tj039cqs44nsu2&prodid=5546%20180

tai ebay:sta

 http://www.ebay.com/itm/SG90-Mini-Gear-Micro-9g-Servo-For-RC-Helicopter-Airplane-Car-Boat-Trex-450-/171417769756?pt=LH_DefaultDomain_0&hash=item27e94b971c

- Sharp Infrared  x2 (Infrapunasensori)
http://www.ebay.com/itm/Sharp-GP2Y0A21YK0F-GP2Y0A21-Infrared-Proximity-Sensor-Distance-Sensor-Perfect-/141587646119?pt=LH_DefaultDomain_0&hash=item20f74806a7

- Mini IR Pyroelectric Infrared PIR Motion Human Sensor Detector Module F5 (liiketunnistin) http://www.ebay.com/itm/Mini-IR-Pyroelectric-Infrared-PIR-Motion-Human-Sensor-Detector-Module-F5-/291391340199?pt=LH_DefaultDomain_2&hash=item43d846fea7

- Tilataanko testaukseen johtimia? Ja mitä muuta roipetta tarvittaan mahollisesti?

Robojuchen, 2015-02-28:
Jos ette halua myöhemmässä elämässä ohjelmoida nalleanne niin voitte ohjelmoida noilla koulun usb-pulikoilla ja tilata itse pelkän pro-minin. 

Servot voi tilata tosiaan ebaysta jolloin suosittelen 1-2extraservon tilaamista sekundakappaleiden varalta tai sitten ostaa kouluelektroniikasta jossa niihin kuulunee takuu.

Teettekö yhden nallen vai molemmille?

Nuo 1.2v oled näytöt on hyviä. Nuo lisämääreet koskevat vain tuommoista multicopteria johon noita yleensä ostetaan