matpi/EmbeddedSystemsUTU #24: 2D-tussipiirturi

Jutossi, 2015-01-19:
Tarkoituksena rakentaa kaksdeetä paperille tussaava piirturi. Voimansiirto toteutetaan kahdella askelmoottorilla ja hammashihnoilla. Tussi lasketaan ja nostetaan käsin. Johteet on 12mm alumiinista pyörötankoa ja kelkka liukuu lineaarilaakereilla johteita pitkin.

Rungon rakenteita oon alotellu ja tilannu osia, mutta voimansiirron kans ei oo tapahtunu mitää. Etenki askelmoottorien(ja mitä muuta niitten lisäksi?) valinnan kans tarttis näkemystä. Unipolar, bipolar!?

Intoa rajattomasti, taitoa rajallisesti, katotaa mite miehen käy?

//Jussi H.

Robojuchen, 2015-01-22:
Oikein hyvä projekti. Katselen vähän että mitkä olisi hyvät motit. Bipolaarisilla yleensä tehdään nämä. Esim nema moottorit ovat suosittuja ja hyviä mutta ei halvimpia. Suosittelisin koneellista kynän nostoa. Ei kait kuvia voi muuten piirtää?

Tietenkin voihan se kone huutaa jatkuvasti että nosta kynä läski...

Jutossi, 2015-01-22:
http://www.ebay.com/itm/Nema17-CNC-Router-Stepper-Motor-2-Phase-4-Wire-1-8-2-5A-Single-Shaft-3D-Printer-/181402150744?pt=LH_DefaultDomain_0&hash=item2a3c692758

Nooh, tallasia kattelin. Osuuko lähellekkään?
Aluksi ajattelin tekeväni yksinkertaisempia kuvia ja mietin manuaalista tussin nostamista, mutta se olisi rajoittava ja ärsyttävä tekijä monimutkaisempien kuvien kans kun joutuu kokoajan nosteleen tussia. Olisiko kynän nostaminen järkevää toteuttaa servolla?

Robojuchen, 2015-01-23:
http://blog.protoneer.co.nz/arduino-cnc-shield/

täältä löytyy valmis shieldi stepperien ajamiseen ja siihen g-koodin kääntäjä kirjasto. Tuon shieldin voi ostaa tai sitten tehdä/tilata kiinasta ilmaisten gerber-tiedostojen pohjalta.

tuohon pitää ostaa tämmöiset yksi per askelmoottori http://www.ebay.com/itm/A4988-StepStick-stepper-driver-for-RAMPS-Pololu-3D-printer-/281537494090?pt=LH_DefaultDomain_0&hash=item418cf13c4a

Robojuchen, 2015-01-26:
Tuo edellinen tapa on siis aika perusteellinen. Käytännössä sillä pystyy tekemään ihan kunnon CNC-koneen. Koodaamista ei tulisi käytännössä ollenkaan koska se on niin valmis paketti. Hintaahan tälle tulee jonkin verran ei voi ihan nollabudjetilla tehdä. Oma juttunsa on sitten laitteen kalibrointi ja noiden tietokonepään ohjelmien säätämien kohdalleen.

Jos haluaa tehdä halvemmalla ja ihan vain piirturin jota ei koskaan päivitä paremmaksi voisi joku tämän tyyppinen lähestymistapa olla hyvä:
http://junkplusarduino.blogspot.fi/p/svg-image-plotter.html
Siinä on perusprinttereistä väännetty aika kiva plotteri ja koodit on jaossa. Tuo arduino-koodi pitäisi sitten vääntää ja kalibroida omille moottoreille sopivaksi.

Robojuchen, 2015-02-11:
http://www.ebay.com/itm/New-A4988-stepper-motor-driver-4-layer-PCB-3D-printer-Pololu-Reprap-Mendel-/281468874242?pt=LH_DefaultDomain_0&hash=item4188da2e02

tuossa olisi suositeltavampi malli noista stepperipiireistä. Huomasin että noissa kaikkein halvimmissa boardeissa virta rajoittuu johonkin 1A tietämille. Nämä olisivat ilmeisesti parempia.

http://www.ebay.com/itm/Nema17-shaft-reversed-stepper-motor-42-BYG-for-RepRap-CNC-machines-3D-Printer-/201266448037?pt=LH_DefaultDomain_0&hash=item2edc6a46a5

jotkut tämmöiset vähän pienemmällä volteilla toimivat moottori olisivat sittenkin ehkä paremmat kuin ne mitä almunperin katselit. Näitä voi ajaa virtarjoituksella ylivolteilla ja saa enemmän nopeutta irti esim.

Robojuchen, 2015-02-11:
http://www.ebay.com/itm/New-Nema17-Stepper-Motor-CNC-Mill-Robot-RepRap-Makerbot-Prusa-Mendel-3D-Printer-/181189450049?pt=LH_DefaultDomain_0&hash=item2a2fbb9941

nämä taitaa olla vähän halvempiakin jotkut nämä pienemmillä volteilla pyörivät mallit.

Jutossi, 2015-02-12:
Okei, hyvä! tänään ajattelin laittaa tilauksia menemää. Jos toteutan sen kynän nostamisen servolla niin sehän ei tarvitse tuota stepperipiiriä? 

tuosta rajakytkimet?

http://www.ebay.com/itm/5pcs-Endstop-mechanical-limit-switch-RAMPS-1-4-for-3D-printer/111362308622?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D1%26asc%3D29171%26meid%3D00beaeadeef34a2f8b1f4824a5af6bb0%26pid%3D100005%26rk%3D2%26rkt%3D6%26sd%3D281235012968&rt=nc

Robojuchen, 2015-02-12:
eli kaksi vaihtoehtoa on nuo GRBL combatible boardit jotka toimii tuon GRBL ohjelmiston kanssa. Sitten nuo halvemmat shieldit vähän vanhempaa mallia toimii vaikkapa tuon junk-projektin koodin kanssa. Jos ei ole tarkoitus koskaan rakentaakkaan laitetta esim. jyrsimeksi tms. niin noilla vanhoillakin (halvemmilla) pärjäilee. 

Ehkä kannattaisi tilata varalta yksi askelmoottori ajuripiiri tuota pystyakselia varten koska nuo vamiit koodit oletuksena nostaa ja laskee askelmoottorilla. Ei tarvitsisi sitten alkaa ihemttelemään miten se vaihdetaan servoksi. Moottoriksi käy joku pieni ja halpakin askelmoottori siihen. 

Katso että noilla myyjillä on yli 99 tuo palaute ja että ne ei ole vielä lähtenyt uudenvuoden lomille :)

Jutossi, 2015-02-14:
Tällä hetkellä tilattuna ja matkalla:
2x askelmoottori
funduino uno
hammashihnaa ja hammasrattaat
rajakytkimet
3x ajuripiirit
GRBL shieldi
johtimia

Jutossi, 2015-02-18:
Vielä uupuu tuo virtalähe tuohon GRBLlle enkä oikee tiiä millanen sen pitäs olla. 

tän tyylisiä näyttää olevan aika paljon vastaavissa projekteissa.

http://www.ebay.com/itm/AC-110V-220V-to-DC-12V-Regulated-Transformer-Power-Supply-For-LED-Strip-Light/361003214629?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D1%26asc%3D29310%26meid%3D87af4b9db05e45179348d706cae5e4ae%26pid%3D100005%26rk%3D2%26rkt%3D6%26mehot%3Dpp%26sd%3D390974992562&rt=nc

eikö perus 12V muuntaja käy?

http://www.kauppasatama.fi/info.asp?tuote=PS12-20A

juhasaario, 2015-03-05:
Jos ei ulkonäkö haittaa, niin ota vanhasta tietsikasta poweri. Yhden romuista pelastin, kun heitelivät syksyllä pois koneita, se on metalliluokan viereisessä varastossa ylähyllyllä. Ota sieltä jos kiinnostaa ja käynnistyksessä saat kyllä apua jos et jaksa ite netittää.