matpi/EmbeddedSystemsUTU #25: "Maaginen Silmä", eli kuka ehtii ensin? (Nunnu)

Nunnu, 2015-01-21:
Ideana on vanha leikki, jossa yksi seisoo selkä pelaajiin päin. Muut yrittävät hiipiä, harppoa tai juosta koskettamaan edessä olevaa ennen kuin tämä ehtii kääntyä katsomaan. Leikki on muunnettu lautapelin muotoon. 

Enintään neljä pelaajaa, jokainen eri puolilla pelialustaa. Keskellä "silmät", jotka aukeavat eri suuntiin ja ovat auki satunaisesti jokaisen pelaajan kohdalla. Pelaajien tulee yrittää saada pelinappulansa liikutettua alustalla keskelle siten, että "silmä" ei liikettä huomaa. Jos silmä kuitenkin huomaa liikken, syttyy valo ja kuuluu ääni sen pelaajan kohdalla joka joutui "narautetuksi". Tämä joutuu palaamaan lähtöruutuun tai muuhun sovittuun kohtaan. Voittaja on se joka pääsee koskettamaan "maagista silmää" keskellä pelilautaa.

Robojuchen, 2015-01-22:
Tämä on kiva idea. Olisiko siinä jokin pieni aika reagoida kun silmä alkaa auteta?

Tähän olisi hyvä liittää vaikka joku näyttö...

Nunnu, 2015-01-22:
Infrapunasensoria (tai jotain sellaista)  kokeilin tänään. Reagoi hyvin käden liikkeeseen kun oli n.5-7 cm korkeudella. Pelinappulan tulee olla matala, jotta "silmä" ei ota matkaa siitä. Arvo ilman kättä oli n. 200+ ja kun käsi tuli "näkökenttään" arvo oli n. 300 ja siitä ylöspäin. Infrapunasensoreja tarttee siis 4 kpl, ainakin yhden summerin, ledejä useampia, ja jokin ( kosketus) katkaisin jokaiselle maaliin päässeelle pelaajalle. Näyttö ei kai ole tarpeellinen?

Mukava olisi myös säätää eri vaikeus- eli nopeustasoja pelille. Jokainen silmä voi olla auki / kiinni eriaikoihin, tai vaikka kaikki auki / kiinni yhtäaikaa. Kunhan pelaajat eivät tiedä milloin mikäkin silmä tuijottaa.

Nunnu, 2015-01-28:
Olen ajatellut, että tarvitsisin seuraavaa

Puuttuu:

Johtoja
Ledit rgb, väh. 4kpl
Infrapunasensorit 4 kpl
Uunon alusta
Vastukset

Löytyy:
Virtalähde, paristot ja 5V muuntaja (muistaakseni, vai oisko 3V)
Pariston pidikkeet
Pietzo, ainakin 1kpl
tavallisia ledejä

Robojuchen, 2015-01-28:
Voisitko laittaa vielä piirroksen miten sensorit ovat sijoitettu ja  onko tarkoitus vain havaita käden liikettä? Mietin vielä että mikä on paras ja halvin sensori tähän laitteeseen. Jos laitteen on tarkoitus toimia paristoilla, voisin suositella lopulliseen asennukseen "arduino pro mini"-boardia niin ei tarvitse jatkuvasti pattereita vaihdella.

Valitkaa sellainen pietzoelementti jolla voi soitella eri korkuisia ääniä niin saatte vähän eloisuutta pelkkään summeriin verrattuna. Tuon näytön tai jokun muun jutun voitte lisätä jos tavoittelette korkeimpia arvosanoja projektista.

Millä ajattelitte aukoa niitä silmiä? Tai miten pelaajalle osoitetaan että hänen silmänsä alkaa juuri katsomaan?

Robojuchen, 2015-01-28:
Laitoin tuohon #13 Leppäkertturyhmän issueen juttua piezosta.

Nunnu, 2015-01-28:
Hei.
Idea on juuri siinä, että pelaaja ei tiedä milloin silmä on auki. Pelaaja joutuu liikkumaan ja olemaan paikalla vain arvaamalla, eli hänen tulee liikuttaa pelinappulaa ihan vaan "säkällä". Kun silmä aukeaa pelaajan liikkuessa, hän joutuu perääntymään jonkin askeleen summerin ja valon syttyessä. Eli käden havaitseminen riittää. Voittaja on se, joka on eniten arvannut kiinni olevia silmiä.

Pelipaikat ovat vastakkain. Malli on pelilauta ja keskellä on neljään ilmansuuntaan olevat sensorit eli silmät. Pelilaudan keskus on se tärkein. Sinne keskelle ajattelin myös sijoittaa kaiken elektroniikan. (Saako tälle alustalle lähetettyä kuvia?)

Pystyykö satunnaista silmän aukaisua ohjelmoimaan, tai sellaista ohjelman kiertoa, jota ei ole mahdollista hoksata ja oppia ulkoa? Nopeutta olisi hyvä pystyä vaihtamaan, sillä kovin hidas pelaaja voi olla hieman hätää kärsimässä.

 Minäpä kattelen Leppäkerttujen kanssa sitä pietzoa.

Tuon arvosanan tavoittelun taidan jättää ja tyytyä näillä resursseilla tähän :)

T. nunnu

Robojuchen, 2015-01-28:
Satunnaisuutta saa arduinosta löytyvällä random komennolla ja kuvia voi näköjään (testi) liittää tähän keskusteluun ihan raahaamalla tähän viesti-ikkunaan.

[image]

Ajattelin että se jonkin lainen silmänavausefekti olisi hyvä jos tavoitteena on peili leikki jossa peili aina kääntyy äkisti ja pelaajilla on minimaalinen reagointiaika kuitenkin+joku psykolginen mahollisuus ennakoida käännöstä. Mutta tämähän on ihan makuasia minkälaisen pelin haluaa, toimii varmasti ilmankin.

Nopeutta voisi säätää potentiometrilla tai sitten laittetta käynnistäessä kädellä noilla etäisyysmittareilla. Ultraäänisensorit voisivat toimia myös ja olla himpun halvemmat, tosin vähän monimutkaisemmat kytkeä ja käyttää. Niillä voisi pelata jopa siten että kävellään kohti laitetta.

Nunnu, 2015-01-29:
Hei. 

Pitäisi päästä kasvotusten ideasta poriseen, mutta olen täällä "Leppäkerttujen " kanssa pureskellut asiaa. Ideaa tuli käyttää kytkimiä pelilaudan " portaina", jolloin se olisi jonkinlainen reaktiopeli. Ääni ja valot ilmoittaisivat kuka pelaaja saa aina siirtyä seuraavalle "reaktiotasolle"...

Robojuchen, 2015-01-30:
katsellaanpa demoilla sitä ideaa. Kuulostaa ihan hyvältä että on jotakin reagointiakin mukana.

Nunnu, 2015-03-24:
Hei! Olen aloitellut ohjelmoinnin raakiletta. Piirilevyllekin olen komponenttien paikkoja katsellut. Kysymys: LCD näytölle on oma output-systeemi. Nyt kun näyttöjä on kaksi, miten ne saa toimimaan eri aikaan, jos ne on kytketty samaan yhdistelmään. Mitä ohjeita olen katsellut, niin olen ymmärtänyt, että Arduinossa on pinnit A4 ja A5 erityisesti näyttöä varten. Miksi muut analogiset eivät käy, vai käyvätkö? Sama dilemma on RGB Ledien kanssa...kaikissa esimerkeissä ne on laitettu pinneihin 9, 10 ja 11. Joku erityinen syykö?

T. Nu

Robojuchen, 2015-03-30:
A4 ja A5 on pinnit I2C dataväylälle. Samoihin pinneihin voidaan kytkeä monta I2C väylää käyttävää laitetta, vaikkapa näyttöä, kunhan niillä kaikilla on eri osoite.

Jos RGB ledejä haluaa käyttää siten että ne saavat portaattomasti eri värejä, värikanavia pitää ajaa pwm-pinnillä. Näitä pinnejä ovat madolla merkityt arduino boardissa, muistaakseni 3,5,6,9, 10 ja 11.

Nunnu, 2015-04-10:
Hei!
Sain tehtyä tätä vielä hieman eteenpäin. Vuokaaviossa on aloitusvalo ennen kuin varsinainen ohjelma alkaa, ja sen paikka jäi hieman epäselväksi. Jos se sijoitetaan setuppiin,kuten ymmärsin,  onko siinä jotain erityistä mitä pitää tehdä? Muuta kommentoitavaa?? Ledit yritin ajastaa paikoilleen...

T. Nunnu

#include <Wire.h> // Tämä tuo I2C-kirjaston käyttöön.

#include <LiquidCrystal_I2C.h> // Ja tämä LCD-kirjaston.

#define BACKLIGHT_PIN 13 // tässä määritellään taustavalopinni

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Asettaa LCD näytön osoitteen 

int led = 3;
int RGBvihr = 9;
int RGBsin = 10;
int RGBpun = 11;
int piezo = 2;
int Silma = A2;
int arvo = 0;
int i = 0;
int liike = 0;
int aiempiArvo = 0;
int taso = 0;

void setup()

{  

  
pinMode (Silma, INPUT);
  
pinMode (RGBvihr, OUTPUT) ;
pinMode (RGBsin, OUTPUT) ;
pinMode (RGBpun, OUTPUT) ;

digitalWrite (RGBvihr,LOW) ;
digitalWrite (RGBsin,LOW) ;
digitalWrite (RGBpun,LOW) ;

pinMode (piezo, OUTPUT);

pinMode (led, OUTPUT);

pinMode ( BACKLIGHT_PIN, OUTPUT ); // asettaa taustavalopinnin OUTPUTiksi
digitalWrite ( BACKLIGHT_PIN, HIGH ); // sytyttää taustavalon
lcd.begin(16,2);               // alustaa lcd-näytön

}

void loop()

{
  
  if (taso < 10)
 
 { 
  digitalWrite (RGBsin,LOW) ;
  delay (3000);
  digitalWrite (RGBsin,HIGH) ;
  delay (100);
  
  liike = 0;   // Tuki onko liikettä
  aiempiArvo = analogRead(Silma)/10 *10; // 
  
  //MITTAA LIIKETTÄ
  
  for(i = 0; i<25; i++) //laskee 15 kertaa loopin läpi, (jos halutaan pitempi aika-suurenna lukua)
{
  arvo = analogRead(Silma)/10 *10; // perästä poistettiin numeroita
  if(abs(arvo-aiempiArvo) > 25) //arvosta vähennetään aiempi arvo 20 kertaa
  {
  liike++;
  }
  aiempiArvo = arvo;  // aiemmastaArvosta tulee arvo, jolloin looppi lähtee alusta
  delay(20);  // kuinka nopeaan silmä lukee liikettä, jos luku pienenee i:tä pitää lisätä
  lcd.setCursor(0,1);         // vaihtaa riviä

  lcd.print(arvo);  
 
}
  i = 0;
  
///LIIKKEEN mittaus loppuu  
  

  lcd.setCursor(0,0);         // vaihtaa riviä
  lcd.print(liike);

  if(liike < 2) // 
  {
  
  digitalWrite (RGBvihr, LOW);
  delay (3000);
  digitalWrite (RGBvihr, HIGH);
  

  lcd.print("Onnea! Palkinto odottaa :)");

  delay(3000);

  lcd.clear();    // tyhjentää näytön
  
  taso = taso + 1;
 
  }

  
  else
  {
  digitalWrite (RGBpun, LOW) ;
  delay (3000);
  digitalWrite (RGBpun, HIGH) ;

  
  pinMode (piezo, OUTPUT);
  delay (3000);
  

  lcd.print(" Harmi! Siirry askel taaksepäin...");      

  delay(3000);

  lcd.clear();
  
  if(taso > 1)
  {
  taso = taso - 1;
  }
  
  }
 
  
  while(taso == 10)
  {
   
  digitalWrite (RGBvihr, LOW);
    delay (3000);
  digitalWrite (RGBpun,LOW) ;
    delay (3000);
    
  pinMode (piezo, OUTPUT);
  delay (3000);
  
  
  lcd.print(" Onnea! Olet voittaja");

  delay(3000);

  lcd.clear();    // tyhjentää näytön
  }
  
}
}

Nunnu, 2015-04-28:
Apua!

En pääse yhtään eteenpäin. Jos yksinäni yritän, minun on luovuttava lcd:stä.Tässä viimeisin yritelmä, mutta en tiedä mitä tehdä:

 
 

#include "pitches.h"

#include <Wire.h> // Tämä tuo I2C-kirjaston käyttöön.

#include <LiquidCrystal_I2C.h> // Ja tämä LCD-kirjaston.

#define BACKLIGHT_PIN 13 // tässä määritellään taustavalopinni

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Asettaa LCD näytön osoitteen 

// notes in the melody:
int melody[] = {
  NOTE_C4, NOTE_G3,NOTE_G3, NOTE_A3, NOTE_G3,0, NOTE_B3, NOTE_C4};

// note durations: 4 = quarter note, 8 = eighth note, etc.:
int noteDurations[] = {
  4, 8, 8, 4,4,4,4,4 };

int led = 3;
int RGBvihr = 9;
int RGBsin = 10;
int RGBpun = 11;
int piezo = 2;
int Silma = A2;
int arvo = 0;
int i = 0;
int liike = 0;
int aiempiArvo = 0;
int taso = 0;

void setup()

{  

pinMode (Silma, INPUT);
  
pinMode (RGBvihr, OUTPUT) ;
pinMode (RGBsin, OUTPUT) ;
pinMode (RGBpun, OUTPUT) ;

digitalWrite (RGBvihr,LOW) ;
digitalWrite (RGBsin,LOW) ;
digitalWrite (RGBpun,LOW) ;

pinMode (piezo, OUTPUT);

pinMode (led, OUTPUT);

pinMode ( BACKLIGHT_PIN, OUTPUT ); // asettaa taustavalopinnin OUTPUTiksi
digitalWrite ( BACKLIGHT_PIN, HIGH ); // sytyttää taustavalon
lcd.begin(16,2);               // alustaa lcd-näytön

}

void loop()

{
  
  
  for (int thisNote = 0; thisNote < 8; thisNote++)   // iterate over the notes of the melody:

  
  {

    // to calculate the note duration, take one second 
    // divided by the note type.
    //e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
    int noteDuration = 1000/noteDurations[thisNote];
    tone(2, melody[thisNote],noteDuration);

    // to distinguish the notes, set a minimum time between them.
    // the note's duration + 30% seems to work well:
    int pauseBetweenNotes = noteDuration * 1.30;
    delay(pauseBetweenNotes);
    // stop the tone playing:
    noTone(2);
  }

 { 
  
  if (taso < 10)
 
 { 
  digitalWrite (RGBsin,LOW) ;
  delay (3000);
  digitalWrite (RGBsin,HIGH) ;
  delay (100);
  
  liike = 0;   // Tuki onko liikettä
  aiempiArvo = analogRead(Silma)/10 *10; // 
  
  //MITTAA LIIKETTÄ
  }
}
  
  
  for(i = 0; i<25; i++) //laskee 15 kertaa loopin läpi, (jos halutaan pitempi aika-suurenna lukua)
{
  arvo = analogRead(Silma)/10 *10; // perästä poistettiin numeroita
  if(abs(arvo-aiempiArvo) > 25) //arvosta vähennetään aiempi arvo 20 kertaa
  {
  liike++;
  }
  aiempiArvo = arvo;  // aiemmastaArvosta tulee arvo, jolloin looppi lähtee alusta
  delay(20);  // kuinka nopeaan silmä lukee liikettä, jos luku pienenee i:tä pitää lisätä
  lcd.setCursor(0,1);         // vaihtaa riviä

  lcd.print(arvo);  
 
}
  i = 0;
  
///LIIKKEEN mittaus loppuu  
  

  lcd.setCursor(0,0);         // vaihtaa riviä
  lcd.print(liike);

  if(liike < 2) // 
  {
  
  digitalWrite (RGBvihr, LOW);
  delay (4000);
  digitalWrite (RGBvihr, HIGH);
 
  
  

  lcd.print("Onnea!)");
  delay(3000);
  lcd.clear();    // tyhjentää näytön

  lcd.print("Palkinto odottaa :)");
  delay(4000);
  lcd.clear();    // tyhjentää näytön
  
  taso = taso + 1;
 
  }

  
  else
  {
  digitalWrite (RGBpun, LOW) ;
  delay (4000);
  digitalWrite (RGBpun, HIGH) ;

 
  
  

  lcd.print(" Harmi!"); 
  delay (3000);
  lcd.clear();    // tyhjentää näytön
  
  lcd.print("Siirry askel taaksepäin..."); 
  delay(4000);
  lcd.clear();
  
  if(taso > 1)
  {
  taso = taso - 1;
  }
  
  }
 
  
  while(taso == 10)
  {
   
  digitalWrite (RGBvihr, LOW);
    delay (3000);
  digitalWrite (RGBpun,LOW) ;
    delay (3000);
 
 
  
  
  lcd.print(" Onnea! Olet voittaja");
  delay(5000);
  lcd.clear();    // tyhjentää näytön
  }
  
}

Robojuchen, 2015-04-29:
Mikä se apua nyt on? Mitä tapahtuu kun koodin ajaa sisään. Mitä ei tapahdu. Mikä on ongelmasi?

Nunnu, 2015-05-08:
Hei!

Olen kokeillut kaikki osaohjelmat erikseen ja valot, ääni ja näyttö toimivat näillä ohjelmilla yksinään ok. Sitten kun ne laitetaan yhteen, sensori lukee liikettä olevan, vaikka mikään tai kukaan ei liiku. Tuntuu, että infrapuna lukee mielivaltaisesti. Kun se lukee jotain??? toiminnot ovat mielestäni kuitenkin oikein. Demoilla kokeiltu kondensaattorikytkentä ei toimi piirilevyllä, sillä se pistää koko näytön "tilttiin". Valot syttyvät ja ääni kuuluu tarvittaessa. Ainoa käytännön ongelma on, että alkaako sensori lukea ja milloin, silloin kun sinisen valo palaa??? vai sen jälkeen??? En osaa ohjelmasta sanoa milloin, sillä tuntuu, että valo sammuu viiveellä ja liikkumisen aloittamista ei siis oikeastaan voi tietää. 

terveisin, nunnu
Alla ohjelma tällä hetkellä:

 
 

#include "pitches.h"

#include <Wire.h> // Tämä tuo I2C-kirjaston käyttöön.

#include <LiquidCrystal_I2C.h> // Ja tämä LCD-kirjaston.

#define BACKLIGHT_PIN 13 // tässä määritellään taustavalopinni

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Asettaa LCD näytön osoitteen 

// notes in the melody:
int melody[] = {
  NOTE_C4, NOTE_G3,NOTE_G3, NOTE_A3, NOTE_G3,0, NOTE_B3, NOTE_C4};

// note durations: 4 = quarter note, 8 = eighth note, etc.:
int noteDurations[] = {
  4, 8, 8, 4,4,4,4,4 };

int RGBvihr = 9;
int RGBsin = 10;
int RGBpun = 11;
int piezo = 2;
int Silma = A2;
int arvo = 0;
int i = 0;
int liike = 0;
int aiempiArvo = 0;
int taso = 0;

void setup()

{  

pinMode (Silma, INPUT);
  
pinMode (RGBvihr, OUTPUT) ;
pinMode (RGBsin, OUTPUT) ;
pinMode (RGBpun, OUTPUT) ;

digitalWrite (RGBvihr,LOW) ;
digitalWrite (RGBsin,LOW) ;
digitalWrite (RGBpun,LOW) ;

pinMode (piezo, OUTPUT);

pinMode ( BACKLIGHT_PIN, OUTPUT ); // asettaa taustavalopinnin OUTPUTiksi
digitalWrite ( BACKLIGHT_PIN, HIGH ); // sytyttää taustavalon
lcd.begin(16,2);               // alustaa lcd-näytön

}

void loop()

{
  

 { 
  
  if (taso < 10)
 
 { 
  digitalWrite (RGBsin,LOW) ; //pelaaja odottaa kunnes valo sammuu
  delay (3000);
  digitalWrite (RGBsin,HIGH) ;
 
  
  liike = 0;   // Tuki onko liikettä
  aiempiArvo = analogRead(Silma)/10 *10; // 
  
  //MITTAA LIIKETTÄ
  }
}
  
  
  for(i = 0; i<25; i++) //laskee 15 kertaa loopin läpi, (jos halutaan pitempi aika-suurenna lukua)
{
  arvo = analogRead(Silma)/10 *10; // perästä poistettiin numeroita
  if(abs(arvo-aiempiArvo) > 25) //arvosta vähennetään aiempi arvo 20 kertaa
  {
  liike++;
  }
  aiempiArvo = arvo;  // aiemmastaArvosta tulee arvo, jolloin looppi lähtee alusta
  delay(20);  // kuinka nopeaan silmä lukee liikettä, jos luku pienenee i:tä pitää lisätä
  lcd.setCursor(0,1);         // vaihtaa riviä

  lcd.print(arvo);  
 
}
  i = 0;
  
///LIIKKEEN mittaus loppuu  
  

  lcd.setCursor(0,0);         // vaihtaa riviä
  lcd.print(liike);

  if(liike < 2) // jos ei ole liikettä
  {
  laulu();
  digitalWrite (RGBvihr, LOW);
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("Onnea!");
  lcd.setCursor(0,1);
  lcd.print("Palkinto odottaa");
  delay(4000);
  lcd.clear();    // tyhjentää näytön
  digitalWrite (RGBvihr, HIGH);
  taso = taso + 1;
 
  }

  
  else
  {

 
  
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("Harmi!"); 
  digitalWrite (RGBpun, LOW) ;
  delay (3000);
  lcd.clear();    // tyhjentää näytön
  lcd.setCursor(0,0);
  lcd.print("Siirry askel");
  lcd.setCursor(0,1);
  lcd.print("taaksepain");
   delay(4000);
digitalWrite (RGBpun, HIGH) ;
 lcd.clear();
  
  if(taso > 1)
  {
  taso = taso - 1;
  }
  
  }
 
  
  while(taso == 10)
  {
   
  digitalWrite (RGBvihr, LOW);
  digitalWrite (RGBpun,LOW) ;
  digitalWrite (RGBsin, LOW);
 
 
  laulu();
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("Onnea!");
  lcd.setCursor(0,1);
  lcd.print("Olet voittaja");
 
  delay(5000);
  lcd.clear();    // tyhjentää näytön
  
  digitalWrite (RGBvihr, HIGH);
  digitalWrite (RGBpun,HIGH) ;
  digitalWrite (RGBsin, HIGH);
  delay(7000);
   
  }
  
}

void laulu()
{
  for (int thisNote = 0; thisNote < 8; thisNote++)   // iterate over the notes of the melody:

  
  {

    // to calculate the note duration, take one second 
    // divided by the note type.
    //e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
    int noteDuration = 1000/noteDurations[thisNote];
    tone(2, melody[thisNote],noteDuration);

    // to distinguish the notes, set a minimum time between them.
    // the note's duration + 30% seems to work well:
    int pauseBetweenNotes = noteDuration * 1.30;
    delay(pauseBetweenNotes);
    // stop the tone playing:
    noTone(2);
  }
}

Robojuchen, 2015-05-11:
Tuossa on ongelmana että heti loopin alussa aaltosulkeita on käytetty miten sattuu. Niillä rajataan vain ohjelman jokin lohko. Eli ne tulevat funktion, loopin, setupin ympärille tai jonkin if lauseen while lauseen jne perään osoittamaan mikä osa koodia suoritetaan sillä ehdolla. ELI NIITÄ EI VOI laittaa muualle.

nyt ohjelmaasa on karkeasti kaksi osiota. Toiseen mennään aina kun taso on pienempi kuin 10 ja toinen johon mennään (while-lause) kun se on 10. Onko tämä tarkoitus. Huomaa että lähes kaikki koodi on nyt ensimmäisen if lauseen sisällä.

#include "pitches.h"

#include // Tämä tuo I2C-kirjaston käyttöön.

#include // Ja tämä LCD-kirjaston.

#define BACKLIGHT_PIN 13 // tässä määritellään taustavalopinni

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE); // Asettaa LCD näytön osoitteen 

// notes in the melody:
 int melody[] = {
 NOTE_C4, NOTE_G3,NOTE_G3, NOTE_A3, NOTE_G3,0, NOTE_B3, NOTE_C4};

// note durations: 4 = quarter note, 8 = eighth note, etc.:
 int noteDurations[] = {
 4, 8, 8, 4,4,4,4,4 };

int RGBvihr = 9;
 int RGBsin = 10;
 int RGBpun = 11;
 int piezo = 2;
 int Silma = A2;
 int arvo = 0;
 int i = 0;
 int liike = 0;
 int aiempiArvo = 0;
 int taso = 0;

void setup()

{ 

pinMode (Silma, INPUT);

pinMode (RGBvihr, OUTPUT) ;
 pinMode (RGBsin, OUTPUT) ;
 pinMode (RGBpun, OUTPUT) ;

digitalWrite (RGBvihr,LOW) ;
 digitalWrite (RGBsin,LOW) ;
 digitalWrite (RGBpun,LOW) ;

pinMode (piezo, OUTPUT);

pinMode ( BACKLIGHT_PIN, OUTPUT ); // asettaa taustavalopinnin OUTPUTiksi
 digitalWrite ( BACKLIGHT_PIN, HIGH ); // sytyttää taustavalon
 lcd.begin(16,2); // alustaa lcd-näytön

}

void loop()

{

if (taso < 10)

{ 
 digitalWrite (RGBsin,LOW) ; //pelaaja odottaa kunnes valo sammuu
 delay (3000);
 digitalWrite (RGBsin,HIGH) ;

liike = 0; // Tuki onko liikettä
 aiempiArvo = analogRead(Silma)/10 *10; // 

//MITTAA LIIKETTÄ
[... 261 lines cut ...]

}

}

void laulu()
 {
 for (int thisNote = 0; thisNote < 8; thisNote++) // iterate over the notes of the melody:

{
// to calculate the note duration, take one second 
// divided by the note type.
//e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
int noteDuration = 1000/noteDurations[thisNote];
tone(2, melody[thisNote],noteDuration);

// to distinguish the notes, set a minimum time between them.
// the note's duration + 30% seems to work well:
int pauseBetweenNotes = noteDuration * 1.30;
delay(pauseBetweenNotes);
// stop the tone playing:
noTone(2);

}
 }

Nunnu, 2015-05-13:
Hei!

Tässä tämä viimeisin ja toimiva ohjelma, T.  nunnu:

#include "pitches.h"

#include <Wire.h> // Tämä tuo I2C-kirjaston käyttöön.

#include <LiquidCrystal_I2C.h> // Ja tämä LCD-kirjaston.

#define BACKLIGHT_PIN 13 // tässä määritellään taustavalopinni

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Asettaa LCD näytön osoitteen 

// notes in the melody:
int melody[] = {
NOTE_C4, NOTE_G3,NOTE_G3, NOTE_A3, NOTE_G3,0, NOTE_B3, NOTE_C4};

// note durations: 4 = quarter note, 8 = eighth note, etc.:
int noteDurations[] = {
4, 8, 8, 4,4,4,4,4 };

int RGBvihr = 9;
int RGBsin = 10;
int RGBpun = 11;
int piezo = 2;
int Silma = A2;
int arvo = 0;
int i = 0;
int liike = 0;
int aiempiArvo = 0;
int taso = 0;

void setup()

{ 

pinMode (Silma, INPUT);

pinMode (RGBvihr, OUTPUT) ;
pinMode (RGBsin, OUTPUT) ;
pinMode (RGBpun, OUTPUT) ;

digitalWrite (RGBvihr,HIGH) ;
digitalWrite (RGBsin,HIGH) ;
digitalWrite (RGBpun,HIGH) ;

pinMode (piezo, OUTPUT);

pinMode ( BACKLIGHT_PIN, OUTPUT ); // asettaa taustavalopinnin OUTPUTiksi
digitalWrite ( BACKLIGHT_PIN, HIGH ); // sytyttää taustavalon
lcd.begin(16,2); // alustaa lcd-näytön

}

void loop()

{

if (taso < 10)

{ 
digitalWrite (RGBsin,LOW) ; //pelaaja odottaa kunnes valo sammuu
delay (3000);
digitalWrite (RGBsin,HIGH) ;

liike = 0; // Tuki onko liikettä
aiempiArvo = analogRead(Silma)/10 *10; // 

//MITTAA LIIKETTÄ

for(i = 0; i<25; i++) //laskee 15 kertaa loopin läpi, (jos halutaan pitempi aika-suurenna lukua)
{
arvo = analogRead(Silma)/10 *10; // perästä poistettiin numeroita
if(abs(arvo-aiempiArvo) > 25) //arvosta vähennetään aiempi arvo 20 kertaa
{
liike++;
}
aiempiArvo = arvo; // aiemmastaArvosta tulee arvo, jolloin looppi lähtee alusta
delay(20); // kuinka nopeaan silmä lukee liikettä, jos luku pienenee i:tä pitää lisätä
lcd.setCursor(0,1); // vaihtaa riviä

lcd.print(arvo); 

}
i = 0;

///LIIKKEEN mittaus loppuu 

lcd.setCursor(0,0); // vaihtaa riviä
lcd.print(liike);

if(liike < 2) // jos ei ole liikettä
{
laulu();
digitalWrite (RGBvihr, LOW);
lcd.clear();
lcd.setCursor(0,0);
lcd.print("Onnea!");
lcd.setCursor(0,1);
lcd.print("Palkinto odottaa");
delay(4000);
lcd.clear(); // tyhjentää näytön
digitalWrite (RGBvihr, HIGH);
taso = taso + 1;

}

else
{

lcd.clear();
lcd.setCursor(0,0);
lcd.print("Harmi!"); 
digitalWrite (RGBpun, LOW) ;
delay (3000);
lcd.clear(); // tyhjentää näytön
lcd.setCursor(0,0);
lcd.print("Siirry askel");
lcd.setCursor(0,1);
lcd.print("taaksepain");
delay(4000);
digitalWrite (RGBpun, HIGH) ;
lcd.clear();

if(taso > 1)
{
taso = taso - 1;
}

}
}

while(taso == 10)
{

digitalWrite (RGBvihr, LOW);
digitalWrite (RGBpun,LOW) ;
digitalWrite (RGBsin, LOW);

laulu();
lcd.clear();
lcd.setCursor(0,0);
lcd.print("Onnea!");
lcd.setCursor(0,1);
lcd.print("Olet voittaja");

delay(5000);
lcd.clear(); // tyhjentää näytön

digitalWrite (RGBvihr, HIGH);
digitalWrite (RGBpun,HIGH) ;
digitalWrite (RGBsin, HIGH);
delay(7000);

}

}

void laulu()
{
for (int thisNote = 0; thisNote < 8; thisNote++) // iterate over the notes of the melody:

{
// to calculate the note duration, take one second 
// divided by the note type.
//e.g. quarter note = 1000 / 4, eighth note = 1000/8, etc.
int noteDuration = 1000/noteDurations[thisNote];
tone(2, melody[thisNote],noteDuration);

// to distinguish the notes, set a minimum time between them.
// the note's duration + 30% seems to work well:
int pauseBetweenNotes = noteDuration * 1.30;
delay(pauseBetweenNotes);
// stop the tone playing:
noTone(2);

}
}