matpi/EmbeddedSystemsUTU #26: Herätysrobotti, osa 2, Juho ja Venla

Venla, 2015-01-22:
Jatketaan Juhon kanssa viime lukuvuonna aloitettua herätysrobotti-projektia. Tavoitteena saada homma taas käyntiin ja avuntarve on välitön. 
Ideana siis tehdä herätysrobotti, joka herättää lähtemällä liikkeelle ja metelöimällä ympäri huonetta, kunnes vastaat oikein sen asettamaan matikanlaskuun. 
Tässä vaiheessa suurimmat haasteet liittyy akkuun ja sen lataukseen sekä näytön ja näppäimistön koodin (joka tuntuu loputtomalta) loppuun saattamiseen. Ääni puuttuu vielä meiltä kokonaan, ja olemme luopumassa ajatuksesta SD-kortilta mp3-tiedostoja soittavasta systeemistä ja tyytymässä ajatukseen valmiista summerista tms. Kommentteja ja ideoita otetaan vastaan!

Robojuchen, 2015-01-22:
Eikun jatkoa vaan. Hyvällä mallilla on teidän projekti. Lataus asiaan oli muistaakseni alkuperäisessä viestiketjussa hahmotella. Äänijutun voi ihan hyvin rajata pois.

Venla, 2015-01-23:
Yritetään päästä omin avuin sille tasolle, missä oltiin keväällä ja tulla sitten aina kun päästään demoille mukaan. 
Nyt tapellaan kellon kanssa. Oliko ajanasettamiseen joku valmis kirjasto? Tällä hetkellä, kello näyttää 00:00 ja serial porttia kun lukee niin päivämää 1.1.2000, kun kokeiltiin aja kirjasto TimeRTCSet. Pitikö sen koodia jotenkin muuttaa, että se hakee ajan vai oliko ihan eri kirjasto vai? Vai tyätyikö kellon joku pinni yhistää arduinon tiettyyn pinnin. Ei yhtään muista.

Robojuchen, 2015-01-23:
Hmmm. Kellon olisi pitänyt olla ajassa jos sen patteri ei ole loppunut. Tuo TimeRTCSet kuulostaa että sillä asetetaan aikaa kellomoduuliin. Mitä sanoo teidän oma kellokoodi jos ajatte sen arduinoon. Teidän kannattaa tarkastaa se kellomoduulin kytkentä, eiköhän netistä löydy oppaita siihen kun haette moduulin nimellä.

Venla, 2015-01-23:
Saatiin toimiin!! 
Ajettiin väärä ohjelma, piti olla se DS1307RTC library ja sieltä Settime example

Venla, 2015-01-23:
Jos yritettäis kaikkein yksinkertaisinta vaihtoehtoa akun kanssa. Eli illalla akku sissään ja robo käyntiin (robotti aukeaa kyljistä), ja aamulla herätyksen jälkeen taas lataukseen, jos tarvis. Mitattiin, että yks servo syö 6mA "lepotilassa"  ja 85 mA ku liikkuu. Muut moduulit vie tietenkin oman virtansa, mutta tuskin mahdottomia määriä. Eli esim. 1000 mAh pitäs riittää hyvin yön yli. Kaikkein helpoin ratkaisu ois, Matinkin kans sitä mietittiin, (vai oisko?) ostaa kouluelektroniikasta valmis paketti ja lyödä vaan sitte lataukseen aina päivällä. mAh on yli 2000, mutta voltteja vaan 4,8, riittääkö? Siis tuo kouluelektroniikka houkuttaa sen takia, että sais tuotteet heti käpäliin, eikä tarvis venailla hongkongista, vaikka maksaakin maltaita. http://www.kouluelektroniikka.fi/cgi-bin/shop.cgi?action=prodshow&usr=mgeb0uareqlbtif3l8bq1f8sa4&prodid=6734%20140

Robojuchen, 2015-01-23:
Toimiihan se tuokin mutta kyllä tuo patterien irrottelu laitteesta tekee siitä käytännössä vähän rasittavan ja epäkäytännöllisen. Jos se ei ole yöllä seinässä niin patteri loppuu nopeasti. Arduino syö 20-30ma servot, 6m, ultraääni 5ma yms. Patterit tyhjenevät yllättävän nopeasti ihan odotellessa herätystä :)

Ettei teidän tarvitsisi tutustua virransäästön kaikkiin hienouksiin, mielestäni tuo aiemmin esittämäni systeemi olisi yksinkertainen ja kätevä. Robotti hyppää telakastaan pois ja patteri kytkeytyy päälle. Kun robotti laitetaan telakkaan lyhyen ajojahdin jälkeen patterien kulutus loppuu saman tien. Niitä ei ts. tarvitse vaihtaa juuri koskaan. Ne voisivat hvyin olla vaikka tavallisia AA pattereita esim 6kpl tai 2kpl li-ion pattereita.