matpi/EmbeddedSystemsUTU #27: elävä taulu

vp-500, 2015-01-22:
Taulu kankaasta jonka taakse elävöitetään kankaan kuvioissa olevia valoja. Löytyy majakkaa, kynttilää autonvaloa jne. Ai niin Jannola odottaa vastausta

Robojuchen, 2015-01-26:
Ajattelitko vielä mikä voisi olla se "älyä" tuova elementti. Lukeeko taulu dataa netistä? Reagoiko se vuodenaikoihin ja vuorokaudenaikoihin? vai tunnistaako se ihmisen edessään ja mahdollisesti tämän liikkeitä?

Jannola, 2015-01-29:
Terve, olen tehnyt sitä rungon kuntoon. Nyt mietin ledien paikkoja ja asennusta ja heijastimia. Tuli mieleen että sensori voisi olla sekä etäisyyttä että ääntä mittaava. Eli kun siinä ihastelee kokonaisuutta niin se innostuu vilkuttamaan kirkkaammin. Jees.

Jannola, 2015-01-29:
Siis olen Petri Viljanen, Annolan kanssa vaan ollaan ja sen nimillä näkyy olevan kone auki