matpi/EmbeddedSystemsUTU #28: Virheilmoitus RFID koodia syötettäessä

Jutossi, 2015-01-22:
Tämä koodi

/**
* Read a card using a mfrc522 reader on your SPI interface
* Pin layout should be as follows (on Arduino Uno):
* MOSI: Pin 11 / ICSP-4
* MISO: Pin 12 / ICSP-1
* SCK: Pin 13 / ISCP-3
* SDA: Pin 10
* RST: Pin 9
*
* Script is based on the script of Miguel Balboa. 
* New cardnumber is printed when card has changed. Only a dot is printed
* if card is the same.
*
* @version 0.1
* @author Henri de Jong
* @since 06-01-2013
*/

#include <SPI.h>
#include <RFID.h>

#define SS_PIN 10
#define RST_PIN 9

RFID rfid(SS_PIN, RST_PIN); 

int buzzPin = 3;

// Setup variables:
    int serNum0;
    int serNum1;
    int serNum2;
    int serNum3;
    int serNum4;

void setup()
{ 
  Serial.begin(9600);
  SPI.begin(); 
  rfid.init();
}

void loop()
{
    
    if (rfid.isCard()) {
        if (rfid.readCardSerial()) {
            if (rfid.serNum[0] != serNum0
                && rfid.serNum[1] != serNum1
                && rfid.serNum[2] != serNum2
                && rfid.serNum[3] != serNum3
                && rfid.serNum[4] != serNum4
            ) {
                /* With a new cardnumber, show it. */
                Serial.println(" ");
                Serial.println("Card found");
                serNum0 = rfid.serNum[0];
                serNum1 = rfid.serNum[1];
                serNum2 = rfid.serNum[2];
                serNum3 = rfid.serNum[3];
                serNum4 = rfid.serNum[4];
               
                //Serial.println(" ");
                Serial.println("Cardnumber:");
                Serial.print("Dec: ");
		Serial.print(rfid.serNum[0],DEC);
                Serial.print(", ");
		Serial.print(rfid.serNum[1],DEC);
                Serial.print(", ");
		Serial.print(rfid.serNum[2],DEC);
                Serial.print(", ");
		Serial.print(rfid.serNum[3],DEC);
                Serial.print(", ");
		Serial.print(rfid.serNum[4],DEC);
                Serial.println(" ");
                        
                Serial.print("Hex: ");
		Serial.print(rfid.serNum[0],HEX);
                Serial.print(", ");
		Serial.print(rfid.serNum[1],HEX);
                Serial.print(", ");
		Serial.print(rfid.serNum[2],HEX);
                Serial.print(", ");
		Serial.print(rfid.serNum[3],HEX);
                Serial.print(", ");
		Serial.print(rfid.serNum[4],HEX);
                Serial.println(" ");
                //buzzer
                analogWrite(3,20);
                delay(500);
                analogWrite(3,0);
             } else {
               /* If we have the same ID, just write a dot. */
               Serial.print(".");
             }
          }
    }
    
    rfid.halt();
}

ANTAA TÄLLAISIA VIRHEILMOITUKSIA... Miss' vika_

RFID\SPI.cpp.o: In function `SPIClass::end()':
C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:40: multiple definition of `SPIClass::end()'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:44: first defined here
RFID\SPI.cpp.o: In function `SPIClass::setBitOrder(unsigned char)':
C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:45: multiple definition of `SPIClass::setBitOrder(unsigned char)'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:49: first defined here
RFID\SPI.cpp.o: In function `SPIClass::setDataMode(unsigned char)':
C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:54: multiple definition of `SPIClass::setDataMode(unsigned char)'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:58: first defined here
RFID\SPI.cpp.o: In function `SPIClass::setClockDivider(unsigned char)':
C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:59: multiple definition of `SPIClass::setClockDivider(unsigned char)'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:63: first defined here
RFID\SPI.cpp.o: In function `SPIClass::begin()':
C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:24: multiple definition of `SPIClass::begin()'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:19: first defined here
RFID\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\RFID/SPI.cpp:40: multiple definition of `SPI'
SPI\SPI.cpp.o:C:\Program Files (x86)\Arduino\libraries\SPI/SPI.cpp:44: first defined here

Robojuchen, 2015-01-22:
tarkastakaa onko arduinon libraries kansiossa RFID-kirjaston kansiossa tiedostot SPI.h ja SPI.cpp ja poistakaa ne jos on

Robojuchen, 2015-01-22:
eli programfiles tms. -> arduino-> libraries -> RFID. sieltä ylimääräiset pois.

matpi, 2015-01-22:
Juu... tuo auttoi. Toinenkaan koodi ei toiminut, ja sieltä löytyi onneksi virheet. Wikin koodihan oli seuraava:

                     #include <SPI.h>
                     #include <RFID.h>
                     #include <LiquidCrystal.h>

                     #define SS_PIN 10
                     #define RST_PIN 9

                     RFID rfid(SS_PIN, RST_PIN); 

                     void setup()
                     { 
                     Serial.begin(9600);
                     SPI.begin(); 
                     rfid.init();

                     }

                     void loop()
                     {

                     if (rfid.isCard()) //onko korttia luettavissa
                     { 
                     if (rfid.readCardSerial()) 
                    {
                    /* Jos kortin lukeminen onnistuu, luvut on tallenettu
                    lukusarjaan josta ne voidaan lukea muuttujiin*/
                    Serial.println(" ");
                    Serial.println("Card found");
                    serNum0 = rfid.serNum[0];
                    serNum1 = rfid.serNum[1];
                    serNum2 = rfid.serNum[2];
                    serNum3 = rfid.serNum[3];
                    serNum4 = rfid.serNum[4];

                    //Serial.println(" ");
                    Serial.println("Cardnumber:");
                    Serial.print("Dec: ");
                    Serial.print(rfid.serNum[0],DEC);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[1],DEC);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[2],DEC);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[3],DEC);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[4],DEC);
                    Serial.println(" ");
                    // tulostetaan sama heksalukuina 
                    Serial.print("Hex: ");
                    Serial.print(rfid.serNum[0],HEX);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[1],HEX);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[2],HEX);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[3],HEX);
                    Serial.print(", ");
                    Serial.print(rfid.serNum[4],HEX);
                    Serial.println(" ");

                    Serial.print(LUETTU!);
                    Serial.println(" ");  
                    delay(1000);
                    }  
                    }

                    rfid.halt();
                    }

Huomasin, että alustuksesta puuttui seuraava pätkä:
    int serNum0;
    int serNum1;
    int serNum2;
    int serNum3;
    int serNum4;

Lisäksi koodiin:
    Serial.print(LUETTU!); piti lisätä hipsut eli näin: Serial.print("LUETTU!");

Nyt Toimii!

Robojuchen, 2015-01-22:
Joo tuli vähän kiireellä tehtyä. Nuo kirjastot pitäisi tarkistaa. Sitä nunchuk kirjastoa ei vissiin ollutkaan muissa kuin siinä yhdessä koneessa jolla kokeiltiin?

matpi, 2015-01-22:
No minulle tämä teki vain hyvää, kun täytyi miettiä mistä kiikastaa :) Mikä sen Nunchuk kirjaston nimi olikaan?

Robojuchen, 2015-01-22:
Se löytyy siltä koneelta millä testailtiin silloin, en muista minkä niminen se on. Hyvin hokasit kyllä nuo lapsukset tuossa koodissa :)