matpi/EmbeddedSystemsUTU #29: Koodiopas

matpi, 2015-01-22:
Johdanto
Tämä materiaali on tarkoitettu muualta opitun tueksi ja ennen opitun kertaamiseksi. Siihen on kirjattu vain arduino-ohjelmoinnin perusasioita, joiden muistelemisesta voi olla kiperän tilanteen sattuessa hyötyä, eikä se näin ollen poista tarvetta opiskella ohjelmointia ruohonjuuritasosta alkaen. Yksinkertaisuudessaan ja monimutkaisuudessaan ohjelmointi on mukavaa aivopähkinää ongelmaratkaisuineen ja motivoivaa työskentelyä näkyvän lopputuloksensa vuoksi.
Serial.println (”Hello World!”);
	-Lauri ja Mikko

        Ohjeita materiaalin käyttöön

Kun ohjelmointia teet ja tämä läpyskä on vieressäsi, istut luultavammin myös internetyhteyden vieressä. Kehotamme lämmöllä käyttämään hakukoneita, sillä niiden hakutulokset usein selventävät sinulle vaikeaksi jäävää asiaa. Materiaaliin on myös tiputeltu ohjelmoinnin komentoja ja termejä helposti erotellussa muodossa, ikään kuin hakuvinkkinä. Ne ovat vaikeita sanoja, jotka on merkitty sulkein ja kursiivilla. Esimerkiksi (muuttuja). Koska Arduinoa kuitenkin ohjelmoidaan C-kielien sukuisella ohjelmointikielellä, voi pelkkä merkitty sana tuottaa hakutuloksia hieman sivuun sinua koskevasta aiheesta, on hyvä lisätä hakuun myös sana ”arduino”. Tarkempaa tietoa saat myös kurssin GitHub-alustalta. Sen wiki-osio on kattava ja yksityiskohtainen, kun tämä vihkonen taas on vain ensihätään.

Esimerkkihaku boolean-muuttujasta:
Google: Boolean

Saat yleislaatuista tietoa boolean-muuttujasta. Toki tämäkin voi olla hyödyllistä, mutta booleaninkin yksityiskohdat muuttuvat koodattavan kielen mukaan.

Google: Boolean arduino

Heti ensimmäinen hakutulos on arduinon oma sivunsa boolean muuttujasta. Tältä sivulta saat varmasti paikkaansa pitävää ja ajantasaista tietoa booleanista ja sen käytöstä.

        Mikä on arduino?

Arduino on ohjelmoitava mikrokontrolleri, eli laite jolla voi tehdä kaikkea kivaa. Sillä voit esimerkiksi vilkuttaa ledejä ja ohjata erilaisia laitteita. Jotkut ovat jopa tehneet älypuhelimella ohjattavan tankin. Yksinkertaisesti sanottuna mikrokontrolleri on mikropiiri, pieni tietokone, jonka sisään- ja ulostuloja voidaan ohjelmoida tietokoneen avulla. 

Arduino koostuu kahdesta osasta: Tietokoneella käytettävästä ohjelmointisovelluksesta ja fyysisestä mikrokontrollerista. Olipa kyseessä mikä tahansa arduino-alusta, sitä ohjelmoidaan samalla sovelluksella ja samalla tavalla kuin muitakin alustoja. Koska Arduino on avoin järjestelmä, laitteita on alkuperäisen Arduino-tuotemerkin lisäksi myös muiden valmistajien tekeminä. Nämä ovat usein kopioita ja paranneltuja versioita alkuperäisistä alustoista. Tunnettuja merkkejä ovat esim. Funduino ja Meduino.

[image]

        Arduino-alustan pinnijaottelu

Kuvassa näkyy Arduino Uno:n tärkeimmät pinnit. Pinnien jaotteluperiaate on sama kaikissa arduinoissa: Digitaaliset sisään- ja ulostulot, analogiset sisääntulot, maapinnit ja jännitteen sisään ja ulostulot. Tarkemmat tiedot löydät aina valmistajan nettisivuilta, tässä tapauksessa http://arduino.cc/en/Main/arduinoBoardUno
 
        Arduino-ohjelmoinnin aloittamiseksi on oltava seuraavat välineet:

o	Arduinotulkki
Arduinotulkki on ohjelma, jonka voi ladata ja asentaa maksuttomasti Arduinon kotisivuilta, Downloads-välilehden takaa osoitteesta: http://arduino.cc/en/Main/Software . Arduinotulkki on mahdollista asentaa sekää Windows, MacOS että Linux-ympäristöihin.
o	Jokin Arduino-mikrokontrolleri
Kurssin puitteissa harjoitusmateriaalina käytetään Arduino Uno ja Arduino Nano –malleja.
o	USB-johto
Sopiva USB-johto tulee yleensä ostetun Arduinon mukana ja koska USB sisältää yleiset liitinstandardit, saattaa sopivia johtoja olla myös muiden laitteiden ohessa.

Kun arduinotulkki on asennettu tietokoneeseen, voi aloittaa koodin tuottamisen vaikka heti. Koodaaminen tapahtuu hieman windowsin muistiota muistuttavassa ikkunassa, johon teksti joko kirjoitetaan näppäimistöllä tai se kopioidaan toiseesta lähteestä.

[image]
 
Kun halutaan siirtää jo valmis koodi tietokoneelta Arduinolle, on ensin kytkettävä Arduino ja tietokone toisiinsa sopivalla USB-johdolla. Kun Arduino on kiinni ja siinä syttyy ledvalo käynnistymisen merkiksi, tulee valita tulkin tools-valikosta kohta ”Board”. Aukeavasta valikosta etsitään omaa Arduinoa vastaava alustamalli. Alustamallin valinnan jälkeen valitaan Serial Port-kohta. USB-väylä on aina merkinnältään COM-alkuinen, esimerkiksi COM24 ja jokainen USB-portti on merkitty eri numerolla. Ensimmäisillä kerroilla tai vieraalla koneella selkeää yhteyttä USB-portin fyysisen paikan ja Serial Port-tunnuksen kanssa ei ole havaittavissa, vaan Serial Port-tunnus on kokeiltava yrityksen ja erehdyksen kautta.

[image]

Kun Arduino on oikealla tavalla yhdistetty tietokoneeseen, on aika kokeilla koodin viemistä tulkilta mikrokontrollerille. Alla olevassa kuvassa näkyy tulkkiohjelman yläreunassa olevat kuvakkeet. 

[image]

Kuvake yksi (1) ajaa koodin tarkastuksen, jossa tulkki tarkastaa koodista esimerkiksi kirjoitusvirheet. Tulkki ei kuitenkaan ymmärrä kaikkia mahdollisia virheitä!
Kuvake kaksi (2) ajaa koodin tulkin läpi ja jos tulkki ei löydä koodista virheitä, ajaa se sen myös mikrokontrollerille.
Kuvake kolme (3) luo uuden tyhjän koodiasiakirjan. Tämän saa tehtyä myös painamalla File ->New tai pikanäppäimellä Ctrl+N.
Kuvake neljä (4) aukaisee aiemmin tallentamasi koodiasiakirjan. Tämän saa tehtyä myös painamalla File -> New tai pikanäppäimellä Ctrl+O.
Kuvake viisi (5) tallentaa aktiivisen koodiasiakirjan myöhempää aukaisua varten. Tämän saa tehtyä myös valitsemalla File -> New tai pikanäppäimellä Ctrl+S.

Jos koodia mikrokontrollerille ajettaessa tulkki antaakin virheilmoituksen, pääsee parhaiten sen sisällöstä selville syöttämällä virheilmoituksen internetin hakukoneeseen. Virheilmoituksia on useita erilaisia ja tässä materiaalissa on käyty läpi vain muutama useimmin ongelmaksi muodostuvista.
Vaikka tulkki olisikin päästänyt koodin lävitseen ja ajanut sen mikrokontrollerille, ei se tarkoita, että koodi toimisi niin kuin sen on tarkoitus. Näissä tapauksissa ei auta kuin koodin läpikäyminen tarkasti. Toimimattoman koodin korjaamiseen, kts. Koodin yleisimmät virheet.
 
        Koodin rakenne

Koodin rakenne näyttää monimutkaiselta, mutta oikeastaan sen toiminta on yksinkertaista. Arduinon ohjelmista löytyvät aina kolme osaa: Määrittelyt, setup ja loop. Arduino-ohjelmassa voi olla myös muita osia, mutta nämä kolme löytyvät aina arduino-ohjelman rakenteesta.

[image]

Esimerkkikuvassa määrittelyt ovat kaksi ensimmäistä riviä. Määritellään int-muuttujat kytkin ja led (kts. Muuttujat). Määrittelyjen jälkeen alkaa setup-osio, jossa pinnit kuusi (kytkin) ja seitsemän (led) määritellään kumpikin omaan käyttöönsä.

Loop-osiossa kerrotaan Arduinolle, mitä tehdä ja missä tilanteessa. Se on osio, jota arduino suorittaa loputtomasti, eli käytännössä niin kauan kuin virtaa riittää. Katso alta visualisoitu kuvio koodin kierrosta.

[... 81 lines cut ...]

Onko sinulla käytössä samat pinnit sekä fyysisesti, että koodissa?

Yksinkertaisimmillaan johdon on kytkenyt eri pinniin, kuin mihin mikropiiri koodin ohjaamana antaa virtaa. Tämä on helppo korjata joko koodista, tai johtoa siirtämällä.

Onko muuttujat kirjoitettu oikein?

Muuttujien pitää olla joka paikassa kirjoitettuna samalla tavalla, että mikropiiri ymmärtää ne oikein suorittaessaan. Katso muuttuja kerrallaan kirjoitusasu ensin määrittelystä ja sitten koodin keskeltä.

Sopivatko muuttujatyypit tehtäväänsä?

Int-muuttujaa ei voi käyttää kirjoitusmerkkiarvona, eikä char-muuttujaa kokonaislukuna.

Onko osilla aikaa suorittaa mikropiirin antama käsky?

Mikropiiri käy koodiloopin läpi sekunnin tuhannesosissa. Tällainen aika ei riitä esimerkiksi servomoottorin liikuttamiseen. Osille pitää antaa toimimisaikaa esimerkiksi delay-komennolla. (Googlaa delay, jos se on sinulle vieras.)

Onko kaikki sulkeet suljettu oikeassa kohtaa?

Sulkeita, lainausmerkkejä ja aaltosulkeita. Tulkki katsoo vain, että näitä on oikea määrä, eikä mikään näistä jää auki. Eli sulkeiden sisällä voi olla käytännössä mitä tahansa, mitä sinne olet kirjoittanut. Seuraa siis sulkeiden avausrivistä aina funktion sulkevaan sulkeeseen, että niiden välissä on oikeat asiat.

Onko koodisi vahingossa kommenttia?

Harvoin, mutta kuitenkin joskus käy niin, että kommentti onkin jäänyt auki ja kaikki sen jälkeinen teksti onkin kommenttia. Vilkaise siis myös, että kommenttisi on lopetettu siihen missä se loppuu.
Jos ongelmaasi ei löytynyt näistä kohdista, käy oman tietotaitosi mukaan koodi läpi, osuiko silmiisi joku virhe? Käy myös läpi funktiosi. Toimiiko if tai while niin kuin meinasit vai pitäisikö sinun koittaa saavuttaa haluttu tulos toisella tavalla? Arduinon oma troubleshoot on myös erittäin laaja ja se löytyy verkosta: http://arduino.cc/en/guide/troubleshooting

Robojuchen, 2015-01-22:
Loistavaa! Noihin virheisiin olisi ehkä hyvä saada ruudunkaapauksia yleisistä koodinkääntötilanteen vikailmoituksista ja selitykset. Näitä on nimittäin vaikea tulkita ja niihin törmätään jatkuvasti.

Sarjaporttia ei välttämättä tarvitse arvailla vaan sen voi katsoa laitehallinnasta (device manager). Tosin usein se on helppo arvata.

Ultraäänianturi ei anna analogista outputtia.

Analogisisia inputteja voi käyttää myös digitaalisina pinneinä, tämä olisi hyvä mainita.

Tästä voisi ehkä tehdä oman sivun wikiin? Voisi sitten muokkailla ja kehittää tätä?

matpi, 2015-01-22:
Juu... Tämä on Laurin ja Mikon tekemä materiaali (viime vuosikurssin opiskelijoita). Se on ollut minulle Pdf- tiedostona, mutta en ole tullut jakaneeksi sitä. Täytyy vielä varmistaa, että pojat eivät pahastu tästä siirrosta tänne wikiin :)