matpi/EmbeddedSystemsUTU #37: Demotehtävä: Dallas lämpötila-anturin kytkeminen Arduinoon

matpi, 2015-02-11:
Oheisesta linkistä löytyy ohje, jonka mukaisesti Dallas lämpötila-anturi kytketään arduinoon.

Kokeilkaa kytkentää. Dallas sensori ei ilmoita lukemaa suoraan asteina... miettikääpä, kuinka saadaan kalibroitua anturi näyttämään asteet tai kuinka löydetään tietyn asteluvun suuruinen raja-arvo anturille? Mitä välineitä tarvitaan?

Yhdistäkää kytkentään LDR vastus. Tehkää ohjelma, joka sytyttää yhden ledin tietyssä valaistuksessa, toisen ledin tietyssä lämpötilassa sekä molemmat ledit tietyn lämpötila ja valoisuusarvon täyttyessä yhtä aikaa.

Robojuchen, 2015-02-12:
Eikös tuo dallas juuri ilmoita ne suoraan asteina? Koodi näkyy paremmin ehjänä ja kommentoituna:
http://milesburton.com/Main_Page?title=Dallas_Temperature_Control_Library

Lämpövastus taas antaa vain muuttuvan analogisen arvon (jännitteenjakokytkennässä).

juhasaario, 2015-03-05:
Nyt ovat mentorit molemmat oikeassa. Dallaspulla ei anna suoraan kuin siansaksaa, mutta kirjasto siistii sillisalaatin selviksi asteiksi. Tosin pieni kupru kirjastoissa on, eivät  välttämättä juttele uusimpien dallaksien kanssa, siinä voi joutua vähän netittämään.

Robojuchen, 2015-03-05:
Jos saivarteluun mennään :) (ja mennäänhän me) niin kyllä tuo dallas antaa ihan selvänä lukemana sen lämpötilan ulos binäärilukuna kunhan osaa onewire protokollalla sen lukea. Eikös kaikki kieli ole siansaksaa jos sitä ei osaa lukea mutta tuolta tulee ulos yksiselitteinen lämpötilalukema etumerkillä ja desimaaleilla höystettynä.

Onko tuosta olemassa jokin uudempi versio kuin DS18B20. Tuo on kait se jota Maxim suosittelee uusiin disaineihin. Tuon sisarmallin DS18B20:n kanssa voi olla ongelmia tässä kirjastossa mutta ei kaít se ole uudempi/parempi malli vaan vähän erilainen vain?

juhasaario, 2015-03-06:
DS18S20 kuulee kyllä komennot, mutta ei välttis juttele takas, semmone etäsuhde tulee sen kanssa.
Ratkaisuna voi tuon kiertää kun käyttää onewireä, tai sitte ei vaan osta sitä ;P

juhasaario, 2015-03-08:
Nonnii juchen... ny kävi sitte nii, että rupes kikkuloiden sielunelämä kiinnostamaan. Olen datasheeteistä aikakaavioita ja registereiden käpistelyä katellu, olihan firmassa logiikka-analysaattori, olihan?
Jos et ole saanut potkuja näin kevättä vasten, niin huvittasko katella kommunikointia ja vaikka yrittää sitä siansaksaa hanskata iha ite, käsin? Saishan siitä sulan hattuun, vaikka siinä mitää järkee oo.