matpi/EmbeddedSystemsUTU #38: KaNu-loukku etätunnistimella

Pietuh, 2015-02-12:
Projektin tarkotuksena tehä perinteinen KaNu-loukku pienpetojen pyyntiin ja siihen tunnistin, joka ilmoittaa milloin loukku on lauennut. Ilmoitus loukun laukeamisesta tulisi tekstiviestinä, sähköpostina tai muuna mahdollisena ilmoituksena. 

Mitä erilaisia mahdollisuuksia tämän toteuttamiseen ois? Kuulostaako liian isotöiseltä?

https://www.google.fi/search?q=kanuloukku&rls=com.microsoft:en-US:IE-Address&source=lnms&tbm=isch&sa=X&ei=cNrcVNnOHsiqywO-64CgCg&ved=0CAgQ_AUoAQ&biw=1280&bih=889

Robojuchen, 2015-02-12:
http://www.ebay.com/itm/SIM900-GPRS-GSM-Board-Quad-Band-Module-For-Arduino-Compatible-High-Quality-New-/331463813989?pt=AU_Gadgets&hash=item4d2cc86b65

Tuossa on halvin löytämäni GSM shieldi arduinolle. Käyttö on varsin yksinkertaista. Tuolla voit vaikka kuunnella kun se supikoira öyhkii loukussa. Jos on tiettyjä vanhoja nokian puhelimia niitä voi käskyttää myös arduinolla ja siten lähettää tekstaria.

Ihan toteutettava ja mielenkiintoinen projekti. Muut kun tekstari/puhelu on vähän hankalampia jos ei olla ihan lähellä kotoa.

Robojuchen, 2015-02-13:
http://www.ebay.com/itm/SIM800L-GPRS-Module-with-PCB-Antenna-Bomb-Slot-Automatic-Micro-

tässä pari aivan uutta tulokasta markkinoille. Piniä ja halpoja. Teoriassa ihan pätevä pulikka, oli pakko tilata yksi koekäyttöön kun on ollut monenlaista automaatioprojektia joita ei ole raaskinut toeuttaa kun on yleensä hieman hintavia nuo GSM moduulit.

Tässä alemmassa on pätevän näköiset antennit mutta toisaalta ei ole mikkiä ja kaiutinliitäntää. Eli pelkkiä viestejä vain voi lähettää...

SIM-Card-/281493146753?pt=LH_DefaultDomain_0&hash=item418a4c8c81
http://www.ebay.com/itm/Smallest-SIM800L-GPRS-GSM-Module-Micro-SIM-Card-Core-Board-Quad-band-TTL-Port-/271765910483?pt=LH_DefaultDomain_0&hash=item3f4682b3d3

Pietuh, 2015-02-16:
okei. kiitoksia! mitkä nokia puhelinmallit ovat niitä joilla tuon voi toteuttaa? ja jos käytän puhelinta ni mitkä osat se silloin korjvaa eli niitä ei tarvii hankkia? ja vanhan nokian puhelimen käytössähän ois varmaan sekin etu että sen akkua ei tarviis ladata  kovin usein ja se on pieni.

Ja tuossa ylimmässä linkissä jos on se kuuntelumahollisuus ni mitä erillisiä lisäosia se vaatii (mikkiä ym.)?

Tuo toinen linkki taas ei toimi ja kolmannesta linkistä avautuu sen verran erilainen osa joka ei varmaa oo se mitä tarkotit?

Mutta mitä kaikkea siis tarvii tuohon työhön?

- tuon gsm shieldin?
- Mikin? ( ja jos tarvii ni milasen?)
- arduinon, muttta millanen ois paras ratkasu tuohon?
- virtalähteen?
- oman sim-kortin?
- piuhoja ja millaislla päillä?
- jotain muuta?

ja millä etäisyydellä tuo toimii? ideaali ois että se lähettäs mulle viestin loukun laukeamisesta vaikka oisin joensuussa käymässä.

Robojuchen, 2015-02-17:
Nyt pitäsi olla nuo linkit kunnossa. Ehkä kannattaisin tuota shieldiä koska nyt ei koko ole kriittinen tekijä. Helppo kytkeä laittamalla suoraan arduinon päälle. Jos tila ja virransäästö ovat oleellisia niin nuo pienet moduulit ovat parempia, mutta uusia juttuja eikä niihin ole noin valmiita oppaita.

Tutki tuon GSM shieldin käyttöä tästä oppaasta:
http://tronixstuff.com/2014/01/08/tutorial-arduino-and-sim900-gsm-modules/

Ja tästä löytyy tarkemmat kuvaukset tuosta shieldistä. 
http://www.geeetech.com/wiki/index.php/Arduino_GPRS_Shield
Siihen voi laittaa mikin tavallisella 3.5mm plugilla. Virtaa tarvitsee sen 2A enimmillään.

Nyt vain suunnittelemaan mitä laite tekee, miten se tunnistaa eri tilanteet. Paljonko se vie virtaa ja kauanko sen pitää toimia? MIkä akku jne. Projektia eteenpäin...

Arduinon valintakin riippuu siitä miten pitkää valmiusaikaa laitteelta vaadit.

Robojuchen, 2015-02-26:
Minulle tuli tuollainen halpa GSM moduuli:
http://www.ebay.com/itm/Smallest-SIM800L-GPRS-GSM-Module-Micro-SIM-Card-Core-Board-Quad-band-TTL-Port-/271765910483?pt=LH_DefaultDomain_0&hash=item3f4682b3d3

Mukana tuli myös antenni ja tuntui onnistuvan tekstarin lähettäminen kun pikaisesti kokeilin DNA prepaid kortilla.

Pietuh, 2015-03-24:
okei nyt pitää tilata nuo osat. on jääny tää projekti tauolle liian pitkäks aikaa muitten projektien takia. mietin että voisin tilata jopa dx.com :ista että tulis osat nopeemmin. Maksaakin enemmän mut ei kuitenka tuu paljoa maksaa muutenka nuo osat. Koska kiinasta tai hongkongista tai muualta sieltä päin tilaa ni luvattukin toimitusaika on pisimmillään vasta huhtikuun lopussa.

Mutta siis tilauslistani ois: 

- tuo shieldi mistä laitoit tuon viimesimmän linkin.
- arduino pro mini 3,3 V
- sit se antenni ja se piuha, minkä nimeä en taaskaan muista, mutta millä saa liitettyä sen isoplugisen antennin pienempiplugiseen arduinoboardiin.

onko oikeat osat vai puuttuuko jotaki oleellista?
nii ja jos meinaan tilata vaikka tuolta dx.com :ista ni ainaka tuollaista vastaavaa shieldiä en löytäny mistä laitoit tuon viimesimmän linkin ebayhin. eli onkoha sielä jotain vastaavaa ees?

Robojuchen, 2015-03-30:
postittaako dx.com nykyään euroopasta? Muuten sieltä ei tule nopeammin. Jätin yliopistolle matin koppiin sen mun gsm moduulin ja voit ottaa sen käyttöön jos tilaat mulle uuden tilalle. 

anenniliitäntä moduulissa on IPEX eli etsi jotain "IPEX SMA adapter" tai  IPEX SMA cable" ja sitten siihen sopiva GSM antenna

Pietuh, 2015-03-31:
joo postittaa minun käsittääkseni myös mutta postittaahan ebaykin näköjään. Deeäxästä en vaa löyt'ny samanlaista shieldiä. Kiitos, voin ottaa ja tilaan sulle samanlaisen.

Mutta mites tuo arduinoboardi? oliko se tuo pro mini 3,3 V mikä minun kannattaa tilata? ja pitääkö siinä olla USB- paikka?

Robojuchen, 2015-04-01:
3.3v ja voit käyttää koulun usb pulikkaa jos et halua itse myöhemmin muuttaa ohjelmistoa.

Pietuh, 2015-04-27:
hei luultavasti aika yksinkertainen ongelma koodissa, mutta en saa siihen ratkaisua vaikka oon aika monesta eri lähteestä yrittäny selvittää. Eli tarviin tuon ongelman ratkaisuun apua.

virheilmoitus on:  unexpected unqualified-id before "{" token
 ohjelma näyttää virheilmotukselle paikaksi tuin ihan ensimmäisen aaltosulkeen kohdassa:

{
        napinTila = digitalRead(buttonPin);
   
   if (napinTila == HIGH || Laskuri > 10799);     //jos nappia painetaan, lähetetään tekstiviesti,  || laitteeseenSoitetaan == true
     
   if (napinTila == LOW) supikoiraMeniLoukkuun = true;
     
   if (Laskuri > 10799); //vuorokausiKulunut
     
   //if laitteeseenSoitetaan = true;
     

   if(supikoiraMeniLoukkuun == true); 
 }

Ja tässä koko koodini tämänhetkisessä muodossa:

 #include "SIM900.h"
#include <SoftwareSerial.h>
//If not used, is better to exclude the HTTP library,
//for RAM saving.
//If your sketch reboots itself proprably you have finished,
//your memory available.
//#include "inetGSM.h"

//If you want to use the Arduino functions to manage SMS, uncomment the lines below.
#include "sms.h"
SMSGSM sms;
//To change pins for Software Serial, use the two lines in GSM.cpp.

int numdata;
boolean started = false;
char smsbuffer[160];
char n[20];
 
 
// **** INCLUDES *****
#include "LowPower.h"
#include <SoftwareSerial.h>
 SoftwareSerial Sarjaportti(7,8);
const int nappiPin = 10;     // napin sijainti
boolean supikoiraMeniLoukkuun = false;
int napinTila = 0;  //napin perustila "painamattomana"?    
int Laskuri ;
int Laskuri2;
int buttonPin = 4;
boolean laitteeseenSoitetaan = true;
boolean vurokausiKulunut = true;
void setup()
{
     //Serial connection.
      pinMode(nappiPin, INPUT); //nappi, the nappi
     Serial.begin(9600);
     Serial.println("GSM Shield testing.");
     //Start configuration of shield with baudrate.
     //For http uses is raccomanded to use 4800 or slower.
     if (gsm.begin(9600)) {
          Serial.println("\nstatus=READY");
          started=true;
     } else Serial.println("\nstatus=IDLE");
}

void loop() 
{
    // Enter power down state for 8 s with ADC and BOD module disabled
    LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);  
   
    // Do something here
    // Example: Read sensor, data logging, data transmission.

}
{
        napinTila = digitalRead(buttonPin);
   
   if (napinTila == HIGH || Laskuri > 10799);     //jos nappia painetaan, lähetetään tekstiviesti,  || laitteeseenSoitetaan == true
     
   if (napinTila == LOW) supikoiraMeniLoukkuun = true;
     
   if (Laskuri > 10799); //vuorokausiKulunut
     
   //if laitteeseenSoitetaan = true;
     

   if(supikoiraMeniLoukkuun == true); 
 }
     {
     lahetaTekstiviesti();
     Serial.print("OK "); //viesti monitorille, että viesti on lähetetty
     exit(0); //lopettaa ohjelman, resetoitava jotta käynnistyy uudelleen
     while(napinTila == HIGH )
     {
       if(Laskuri2 < 7200)
       {
        lahetaTekstiviesti();  
     }
      LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
      Laskuri++;
      Laskuri2++;
       napinTila = digitalRead(buttonPin);
     
   }
     }
   
   
  if(vuorokausiKulunut == true) {
    //soitaHaly();
    vuorokausiKulunut = false;
    Laskuri = 0;
  }
  
   // if(laitteeseenSoitetaan == true) {
   // vastaaSoittoon();
  //}
 
 //kerrotaan iffissä esitelty lahetatekstiviesti "metodi" joka lähettää tekstarin nroon.
 void lahetaTekstiviesti {
  
  if(started) {
          Enable this two lines if you want to send an SMS.
         if (sms.SendSMS("3471234567", "Arduino SMS"))
          Serial.println("\nSMS sent OK");
     } 
 }

Pietuh, 2015-05-05:
Tässä viestissä kaikista uusin versio koodistani kokonaisuudessaan. Nyt eri virheilmoituksella, joka on: "expected constructor, destructor, or type conversion before "=" token" ja tuo virhe ilmoitetaan tähän kohtaan: "   napinTila = digitalRead(buttonPin)"

 #include "SIM900.h"
#include <SoftwareSerial.h>
//If not used, is better to exclude the HTTP library,
//for RAM saving.
//If your sketch reboots itself proprably you have finished,
//your memory available.
//#include "inetGSM.h"

//If you want to use the Arduino functions to manage SMS, uncomment the lines below.
#include "sms.h"
SMSGSM sms;
//To change pins for Software Serial, use the two lines in GSM.cpp.

int numdata;
boolean started = false;
char smsbuffer[160];
char n[20];
 
 
// **** INCLUDES *****
#include "LowPower.h"
#include <SoftwareSerial.h>
 SoftwareSerial Sarjaportti(7,8);
const int nappiPin = 10;     // napin sijainti
boolean supikoiraMeniLoukkuun = false;
int napinTila = 0;  //napin perustila "painamattomana"?    
int Laskuri ;
int Laskuri2;
int buttonPin = 4;
boolean laitteeseenSoitetaan = true;
boolean vuorokausiKulunut = true;
void setup()
{
     //Serial connection.
      pinMode(nappiPin, INPUT); //nappi, the nappi
     Serial.begin(9600);
     Serial.println("GSM Shield testing.");
     //Start configuration of shield with baudrate.
     //For http uses is raccomanded to use 4800 or slower.
     if (gsm.begin(9600)) 
     {
          Serial.println("\nstatus=READY");
          started=true;
     } else Serial.println("\nstatus=IDLE");
}

void loop() 
{
    // Enter power down state for 8 s with ADC and BOD module disabled
    LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);  
   
    // Do something here
    // Example: Read sensor, data logging, data transmission.

}

   napinTila = digitalRead(buttonPin)
   
   if (napinTila == HIGH || Laskuri > 10799)//jos nappia painetaan, lähetetään tekstiviesti,  || laitteeseenSoitetaan == true
     
   if (napinTila == LOW) supikoiraMeniLoukkuun = true
     
   if (Laskuri > 10799)//vuorokausiKulunut
     
   //if laitteeseenSoitetaan = true;
   

   if (supikoiraMeniLoukkuun == true) 
     {
     lahetaTekstiviesti();
     Serial.print("OK "); //viesti monitorille, että viesti on lähetetty
     exit(0); //lopettaa ohjelman, resetoitava jotta käynnistyy uudelleen
     while(napinTila == HIGH )
     {
       if(Laskuri2 < 7200)
       {
        lahetaTekstiviesti();  
     }
      LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
      Laskuri++;
      Laskuri2++;
       napinTila = digitalRead(buttonPin);
     
   }
     }
   
   
  if(vuorokausiKulunut == true) {
    //soitaHaly();
    vuorokausiKulunut = false;
    Laskuri = 0;
  }
  
   // if(laitteeseenSoitetaan == true) {
   // vastaaSoittoon();
  //}
 
 //kerrotaan iffissä esitelty lahetatekstiviesti "metodi" joka lähettää tekstarin nroon.
 void lahetaTekstiviesti {
  
  if(started) {
          Enable this two lines if you want to send an SMS.
         if (sms.SendSMS("3471234567", "Arduino SMS"))
          Serial.println("\nSMS sent OK");
     } 
 }

Robojuchen, 2015-05-05:
laitan sulle sähkpostiin korjaillun koodin. Tuossa on nuo sulkeet vähän päin mäntyä. Ohjemassa on aina seuraava rakenne:

void loop()
{
tässä kaikki koodi lukuun ottamatta ali ohjelmia

}

elikkä sulla olisi tällaisia if lauseita siellä sitten:

void loop()
{
loopissa olevat rivit toteutuu aina kun ohjelma pyörii mutta...

if (jokin on totta)
{
kaikki mitä tässä välissä on tapahtuu jos ehto täytyy
}

}

lisäksi if-lausee ehdossa ei ole puolipistettä mutta komennoissa kun jokin muuttuja laitetaan true vaikkapa pitää aina olla puolipiste.

sinun ohjelman idea:

1.nuku
2.tarkista nappi
3. Lisää laskurin arvoa
3.jos nappi on painettu tai laskuri on täyttytynyt
a) laita muuttujat trueksi sen mukaan mitä on tapahtunut
b) jos supikoira on mennyt loukkuun
-lähetä tekstari
- niin kauan kun nappi on painettuna tarkista onko laskuri 2 täyttynyt
- jos on niin lähetä uusi tekstiviesti ja nolla laskuri2
-muuten nuku aina 8s ja lisää laskuri2 arvoa ja lisää myös laskurin arvoa jotta yleiskello pysyy ajassa.

c)jos nappi vapautuu tai alunperinkään supikoiraa ei ole loukussa vaan aika laskuri on täyttynyt
-tarkista onko laskuri täyttynyt
-jos on, soita häly ja nollaa laskuri

sitten alusta uudestaan

HUOM katso että hahmotat tämän logiikan koodista!!!

Robojuchen, 2015-05-05:
kokeile saisitko tällä toimimaan kun en jaksa selvittää s-postia: tämä github hukkaa nuo included osittain....

#include "SIM900.h"
 #include <SoftwareSerial.h>
 //If not used, is better to exclude the HTTP library,
 //for RAM saving.
 //If your sketch reboots itself proprably you have finished,
 //your memory available.
 //#include "inetGSM.h"

//If you want to use the Arduino functions to manage SMS, uncomment the lines below.
 #include "sms.h"
 SMSGSM sms;
 //To change pins for Software Serial, use the two lines in GSM.cpp.

int numdata;
 boolean started = false;
 char smsbuffer[160];
 char n[20];

// **** INCLUDES *****
 #include "LowPower.h"
 //#include 
 SoftwareSerial Sarjaportti(7,8);
 const int nappiPin = 10; // napin sijainti
 boolean supikoiraMeniLoukkuun = false;
 int napinTila = 0; //napin perustila "painamattomana"?

 int Laskuri ;
 int Laskuri2;
 int buttonPin = 4;
 boolean laitteeseenSoitetaan = false;
 boolean vuorokausiKulunut = false;
 void setup()
 {
 //Serial connection.
 pinMode(nappiPin, INPUT); //nappi, the nappi
 Serial.begin(9600);
 Serial.println("GSM Shield testing.");
 //Start configuration of shield with baudrate.
 //For http uses is raccomanded to use 4800 or slower.
 if (gsm.begin(9600)) 
 {
 Serial.println("\nstatus=READY");
 started=true;
 } else Serial.println("\nstatus=IDLE");
 }

void loop() 
 {
 // Enter power down state for 8 s with ADC and BOD module disabled
 LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF); 
// Do something here
// Example: Read sensor, data logging, data transmission.

Laskuri++;

napinTila = digitalRead(buttonPin);

if (napinTila == HIGH || Laskuri > 10799)//jos nappia painetaan, lähetetään tekstiviesti, || laitteeseenSoitetaan == true
{
if (napinTila == LOW) 
supikoiraMeniLoukkuun = true;

if (Laskuri > 10799)//vuorokausiKulunut
{
vuorokausiKulunut = true;
}

if (supikoiraMeniLoukkuun == true) 
 {
 lahetaTekstiviesti();
 Serial.print("OK "); //viesti monitorille, että viesti on lähetetty
 //exit(0); //lopettaa ohjelman, resetoitava jotta käynnistyy uudelleen
 while(napinTila == HIGH )
 {
 if(Laskuri2 < 7200)
 {
 lahetaTekstiviesti();
 Laskuri2 = 0;

 }
 LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
 Laskuri++;
 Laskuri2++;
 napinTila = digitalRead(buttonPin);

}
 }

supikoiraMeniLoukkuun = false;

if(vuorokausiKulunut == true) {
 //soitaHaly();
 vuorokausiKulunut = false;
 Laskuri = 0;
 }
}
// if(laitteeseenSoitetaan == true) {
 // vastaaSoittoon();
 //}
 }
 
 
 
 
 
//kerrotaan iffissä esitelty lahetatekstiviesti "metodi" joka lähettää tekstarin nroon.
 void lahetaTekstiviesti() {

if(started) {
 //Enable this two lines if you want to send an SMS.
 if (sms.SendSMS("3471234567", "Arduino SMS"))
 Serial.println("\nSMS sent OK");
 } 
 }

Pietuh, 2015-05-10:
Hei ootko löytäny sitä koodinpätkää netistä sihen minun koodiin? eli sen laitteen pitäs soittaa mulle vuorokauden välein häly varmistaakseen että se toimii. Toki saman asian ajaisi se että laite lähettäisi mulle viestin vuorokauden välein. Kumpi metodi vaan on helpompi toteuttaa ja istuttaa tuohon minun koodiin. Minä lähen nyt ens keskiviikkona kesälomille ja mulla alkaa työt, mutta mulla on tavoitteena että saisin tuon arduinoprojektin valmiiksi siihen mennessä ja kurssin suorittanminen ei jäisi roikkumaan ensi vuotten, jolloin luultavasti en muista siitä edes sitä vähää mitä nyt tajuan. Ja korostan vielä että mulla ei todellakaan oo tavoitteena lähteä hakemaan kuuta taivaalta, vaan tärkein asia mulle on saada tuo projekti valmiiksi viimeistään keskiviikoksi. Siinäkin on mulle ja miun tiedoille ja taidoille arduinohommissa tarpeeksi työsarkaa.

Robojuchen, 2015-05-11:
siinä kirjastossa oli valmis toiminto sille
soittaa puhelun
gsm.Call("+358123456789");

lopettaa puhelun
 call.HangUp();

 huomenna ja ke aamuna voidaan kattoa hommaa vaikka paikan päällä jos ongelmaa tulee

Robojuchen, 2015-05-11:
http://www.headele.com/Datasheet/Wireless%20module/GPRS/SIMCOM/SIM800L_Hardware_Design_V1.00.pdf

Sivulla 31 kuva miten tx ja rx kytketään.  Muita datalinjoja ei tarvitse kytkeä Moduulin käyttöjännitteeksi tuo 3.4v -4.4 v eli vaikkapa kännykän akku tai vastaava suoraan moduulin poweriin tai diodi siihen väliin.

Robojuchen, 2015-05-11:
tuo GSM moduuli hörppii aika lailla virtaa jos sitä ei nukuteta siksi ajaksi kun sitä ei käytetä. Eli jos akkusi ei ole valtavan iso niin sille asialle olisi hyvä tehdä jotakin.

sen voi nukuttaa komennolla
gsm.SimpleWriteln("AT+CPOWD=1");

ja herätys tapahtuu että moduulin rst pinniin kytketty arduinon pinni laitetaan low-tilaan vähän yli 100ms ajaksi.

TOINEN VAIHTOEHTO:

Koko moduulin virta kytketään pois ja päälle tarpeen mukaan sen  ground-johtoon laitetulla N-fetillä. Tällä saavutetaan suurin virransäästö.